keirf/flashfloppy #758: Ensoniq asr 10 cant swapping images

grandpuba22, 2023-02-13:
Hello. I have a problem saving the project to 2 discs ensoniq asr10 and akai mpc3000. The flash floppy emulator does not understand changing the image disk. I tried many options in changing the file cfg. settings)cant save to multiple floppies.

grandpuba22, 2023-02-15:
Anybody has good images for ensoniq asr10 and good config file?to save project with multiple floppies

keirf, 2023-02-18:
Usually disk change problems are IBMPC vs Shugart interface settings. Have you tried changing `interface=` setting in FF.CFG? And/or adding jumper JC at rear of Gotek?

grandpuba22, 2023-02-18:
I am changing interface ensoniq asr…nothing happens)shugart and ibmpc not hellping)ensoniq asr works only with ibmpbc-hdout)

grandpuba22, 2023-02-19:
My drive is eilasung fdd-udd u144k)with ensoniq asr10 it’s only works with hbmpc-hdout host ensoniq.I tried change inteface to shugart or ibmpc it,s not working)I dont know where the jumper jc?

keirf, 2023-02-19:
You should ask in the discussion area in GitHub or in the FlashFloppy Facebook group. This area is for firmware bugs rather than general support.

keirf, 2023-02-19:
One last thing for you to try is chgrst=delay-3 in FF.CFG

closed 2023-02-19 by keirf

grandpuba22, 2023-02-19:
Thanx i will try