keirf/flashfloppy #760: Roland s760 gotek c'ant communicate message!

shindoja, 2023-02-15:
Hi ! A have a s760 with a Gotek booting fine with 000.IMG :)
But i cannot load samples .IMG file its said "cant communicate" on the s760 and error 31 on the Gotek.
What im i missing ? Thanks a lot

tenuke, 2023-02-15:
https://github.com/keirf/FlashFloppy/wiki/Error-Messages

error 31 = Bad or Unsupported image file

shindoja, 2023-02-15:
Yep a saw that ! But still ? Its the .IMG a found on that link : https://onedrive.live.com/?authkey=%21AFBw7J%2D2SF9Fr30&id=D541D8022D453E0C%211749&cid=D541D8022D453E0C&parId=root&parQt=sharedby&parCid=516454F7667BB874&o=OneUp

And : http://llamamusic.com/s50s550/unusual_samples.html

tenuke, 2023-02-15:
> Yes im not saying that the files are broken, just unsupported by
> flashfloppy firmware.
>
> Did check that package from onedrive link, the file extensions are .OUT
> That another link did show some tools which you should check and see if
> you can convert those .out to something like .img

oh the .OUT is listed as an supported format but anyways, because the FF
gives you this error, I would still go forward by trying
to convert the images :o

shindoja, 2023-02-15:
Ok ! Still doesnt work hehe
I changed the extension to IMG ang to OUT. 
I update to 4.7 but blanc screen after the countdown
I came back to 3.38 and it booting again
But when i try to load its doesnt see and say : Please insert disk 

The thing is im a musician haha I dont know much about computer or even de 760 as i just got it

So really maybe there is something really stupid a dont get hehe

Do i have to make a special partition on my usb key for s760 to read the sounds ?

And also , on the Gotek, There is only 4 " parts " a can click on : 000, 001, 002, 003 

Thanks you so much !

keirf, 2023-02-18:
Can you share a non-working IMG file on this issue ticket?

shindoja, 2023-02-20:
[image]

shindoja, 2023-02-20:
Oups it says that

shindoja, 2023-02-20:
[image]

I just rename it Jpeg to paste it here, but its the IMG file that i tried to play in the S760 :) Thank you

tenuke, 2023-02-20:
Try to make an zip-package of it first, then drag&drop that .zip here :)
WinZip for example works for zipping

shindoja, 2023-02-20:
[002.IMG.zip](https://github.com/keirf/flashfloppy/files/10787349/002.IMG.zip)

shindoja, 2023-02-20:
Thanks for the tip !

shindoja, 2023-03-08:
Anybody has a clue ? When i go to dick it say " please insert disk " , Its like i would have to make a partition or something , my usb key is 16g
Thank you !

keirf, 2023-03-09:
IMG file looks fine.

Maybe you need to create a FF.CFG text file in your USB stick with a different interface setting in it. It's just a plain text file. Eg you could try putting this in it:
```
interface = jppc-hdout
```

shindoja, 2023-03-09:
Ok i'll try that ! Thanks :)

shindoja, 2023-03-09:
Oups does not boot anymore. And i tried to revert to shugart and does not boot either. There is definitely something i dont get in the general structure of the Gotek protocoles hehe. I did eventually push the 2 buttons to try reseting..may be i took it back to stock parameters ? 

So from the start what i understand: 

1- Unbox the gotek
2- Put the J3 jumpers to install firmware
3- Use artery to install firmware into Gotek via usb to usb cable
4- Put Gotek physically into the s760
5- Format usb stick in fat 32 ( 16gig)
6- Put s760224 file from roland on the root folder 

That's how i got to boot the s760 until now but never was able to save or read a sample.

Thank you ! Cant wait to play music haha

shindoja, 2023-03-19:
Could it be that im installing the wrong hex file in the Gotek ? The file installed is flashfloppy t415-st105-3.38.hex 

Im trying to update to a HxCFEUSB_5_1_1_1aRoland_QuickDisk.UPD but the Gotek wont update nothing it seems.

shindoja, 2023-03-19:
Ok so looks like I have to put a resistor on pin 2 and 34 hehe come back with details !

shindoja, 2023-03-19:
Is it normal also it does not keep the .CFG file ?

keirf, 2023-03-19:
Which CFG file?

shindoja, 2023-03-19:
Also I did the resistor thing on pins 2 and 34 and it did not change nothing hehe

shindoja, 2023-03-19:
Also its like the gotek does not want to update files when I put them on the USB stick and hold the 2 buttons at start. Its says upd but different error code after that.

keirf, 2023-03-19:
The config file you write to should be called FF.CFG. Are you writing to IMAGE_A.CFG?

shindoja, 2023-03-19:
A yes as it was the .CFG file that the s760 created i though its was the right one !
Ok so i'll try whit the FF one. Also : 

1- if you could tell me if I should let the resistors or remove them that would help remove some shaky parameters from the overall situation :)

2- What firmware should i put to be sure now i flashed the : at415-st105-3.39.hex

3- What could be the issue with the update protocol ?

Thanks infinitely

keirf, 2023-03-19:
I wouldn't mess with the onboard resistors.

What model is your Gotek (written on the PCB)?

shindoja, 2023-03-19:
ok ! 

Model : SFRKC30.AT2

Also it only boot with : interface = jppc-hdout or IBM PC

shindoja, 2023-03-19:
Also we are talking on facebook haha

keirf, 2023-03-19:
Do you have OLED display? What version is shown when you enter update mode?

shindoja, 2023-03-19:
No oled just stock version ( plan to do that later )

keirf, 2023-03-19:
You have flashed the correct firmware. Update should work. What's the error.

shindoja, 2023-03-19:
Most of the time 34 sometimes 01

keirf, 2023-03-19:
E01 is no update file found. E34 I think shouldn't be possible in update mode.

keirf, 2023-03-19:
You should copy the .upd file from the Greaseweazle distribution to root of your usb stick.

shindoja, 2023-03-19:
Says e56 for 1 sec then e01

keirf, 2023-03-19:
What file(s) did you copy to your USB drive?

shindoja, 2023-03-19:
[greaseweazle-firmware-1.3.upd.zip](https://github.com/keirf/flashfloppy/files/11012427/greaseweazle-firmware-1.3.upd.zip)

shindoja, 2023-03-19:
And i also tried to flash the gotek with the .hex file but the gotek did not lite up after. Illl try the f1 and f7 version

shindoja, 2023-03-19:
[image]

keirf, 2023-03-19:
Why are you using Greaseweazle firmware files? You want FlashFloppy right?

shindoja, 2023-03-19:
Those are the files on the root of usb stick 000 is the roland 2.24

shindoja, 2023-03-19:
Is the stick should be in masterboot or partition guid ?

keirf, 2023-03-19:
Recent firmware supports both mbr and gpt.

shindoja, 2023-03-19:
ok

keirf, 2023-03-19:
I've heard no success loading s760 samples. What would be useful is a s760 sample disk layout. Eg an hfe image file formatted on s760. Or areal sample disk dumped using something like Greaseweazle. Then we could see the actual disk layout.

shindoja, 2023-03-19:
None of the greaseweaszle work on my Gotek :(

shindoja, 2023-03-19:
I mean the firmware

keirf, 2023-03-19:
Of course not. Greaseweazle doesn't run on Gotek.

shindoja, 2023-03-19:
I dont have any original disk

keirf, 2023-03-19:
Let me ask around.

shindoja, 2023-03-19:
Ok thanks a lot !

shindoja, 2023-03-20:
Its funny cause the s760 always re-create a new IMAGE_A.CFG file on the usb stick each time a boot it with it.

keirf, 2023-03-20:
Yes it uses that to remember last image loaded.

shindoja, 2023-03-20:
There's developpement ! I put the pin 2 and 34 in the ff.CFG file and it loaded 1 .IMG file but says : please insert disk 2 and then er and i have to cancel.

Also the other disk image say : ' unformated ' and ' not a sound disk '. I tried to format but change nothing.

This is my ff.CFG : 

 host = unspecified
 interface = jppc-hdout
 pin02 = rdy
 pin34 = chg

I'll try ibmpc-hdout  :) At least i know everything it good physically with the Gotek!

shindoja, 2023-03-20:
[002.IMG.zip](https://github.com/keirf/flashfloppy/files/11018165/002.IMG.zip)
[DISK3.IMG.zip](https://github.com/keirf/flashfloppy/files/11018166/DISK3.IMG.zip)

The DISK3.IMG load but say error at the end ( feels like its too big or something )
The 002.IMG it say ' not sound disk '

shindoja, 2023-03-20:
I'll try dens on pin 2

daftpunker, 2023-03-22:
Hi all, I'm trying to save a sample from s760 on the gotek image senza successo. The gotek is correctly installed (jumper setting on JC and S0) in fact the OS on the s760 loads correctly, the problem occurs during the saving phase, first it asks to format the disk, then it starts writing, but in the end either a "floppy write error" error arrives or in other cases it hangs loading and crashing, so I'm forced to turn the s760 off and on again. I have gotek on roland S760 and on Roland dj70 are very similar, in fact I have the same problem with both. The s760 gotek has firmware 3.35 The dj70 gotek has firmware e 3.36. I enclose images of the Roland S760, the tests were done both on image 1.44 and 720 and screen of the usb stick and folders and images. I also tried 3 different stick usb with the same result. Can someone help me? 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

daftpunker, 2023-03-22:
https://youtu.be/-MvFgu-2ktM

https://youtu.be/nnP3Vx2FgEk

cidermassive, 2023-03-22:
Can you load these images into your DJ70? 
[DJ70-DemoDisks.zip](https://github.com/keirf/flashfloppy/files/11045021/DJ70-DemoDisks.zip)

If not, let me know and I will see what settings I have in my FF.CFG

daftpunker, 2023-03-24:
Hi cidermassive, thanks but it didn't work, it won't load the demos you sent me.

daftpunker, 2023-03-24:
keirf, thanks to you too for the answer, on the s760 it only loads the operating system without any problem. Then I try to export by saving a sample on gotek but after waiting for the usual formatting of the blank and started writing it returns at the end of the error

daftpunker, 2023-03-24:
I have made various tests, I am attaching 2 zip folders that precisely reproduce the contents of the 2 USB sticks  s760 and dj70
[S760.zip](https://github.com/keirf/flashfloppy/files/11066012/S760.zip)
[DJ70.zip](https://github.com/keirf/flashfloppy/files/11066013/DJ70.zip)

[43 more replies]

closed 2023-06-27 by keirf