keirf/flashfloppy #761: Cyls > 80 problem

kivijakola, 2023-02-16:
I'm trying to write to a raw .img image that has following format:

cyls = 83
heads = 2
secs = 10
bps = 1024

I have empty .img file stored on usb stick and writing is working fine up to the cylinder 80. After that I always get Windows message "The drive cannot find the sector requested" Writing is working fine with standard floppy drive and diskette.

There is not any error message in uart output either. I couldn't spot the place in source code that is triggering this error. 

I'm using SAMdisk for writing: SAMdisk.exe a: hunter.dsk -s10 -z3

Edit:
Seems that this print is always coming out when cyl index >= 80:
                printk("IMG Bad CRC: %04x, %u[%02x]\n",
                       crc, sec_nr, sec->r);

SAMdisk is returning error if cylinder is exactly 80 but there is not error displayed when writing single sector higher than 80 but CRC error is still there and data is not written.

Could that be a generic issue when cylinder > 80?

kivijakola, 2023-02-16:
Code in img.c looks like doing some very weird. Can't imagine what it should...

There is for example definition at line 2020: `unsigned int nr, todo, sec_sz;`
Then nr is used uninitialized at line 2066: `for (todo = sec_sz; todo != 0; todo -= nr) {`

keirf, 2023-02-17:
Is the empty .img file the correct size? `83*2*10*1kB = 1660kB = 1699840 bytes`

keirf, 2023-02-17:
Another option is that these tracks >= 80 have deliberately bad CRC. Could be copy protection for example. Or perhaps the tracks aren't really used.

If you can share the EDSK image file I can have a look.

Also you could try writing to a double-density HFE image file. There's one here: https://github.com/keirf/flashfloppy-images/blob/master/Unformatted/HFE/double_density.hfe

kivijakola, 2023-02-18:
Image for testing is available here: https://kivijakola.fi/share/hunter30.dsk. It may be true that those >=80 tracks are not used but SAMdisk is for some reason reading them. Ordinary floppy drive does not care any CRC but flashfloppy is doing checkin. I don't know is that good or bad thing.

Anyways, I got my Hunter alingment machine booting with Gotek and flashfloppy! Reason wasn't those extra tracks. Just skipped them. Actual problem was that interface = jc was not correct even that it was working with PC. I had to set is interface = ibmpc.

BTW: copying hunter30.dsk directly to usb stick was not working either. SAMdisk returned half of size image when trying to read it back. But raw .img is now enough and working!

keirf, 2023-06-27:
Marked as bug just so that I remember to check the behaviour of `hunter30.dsk` myself. I will then fix if necessary, and then close the ticket.

keirf, 2023-07-04:
The issue is that hunter30.dsk is a high density EDSK image. Neither FlashFloppy nor Greaseweazle currently supports that. I can probably fix it.