keirf/flashfloppy #763: About : No serial flow control on Z80-MBC2

eprive, 2023-02-19:
You may have heard about a newer branch of the Z80-MBC2 project called : Z80-MBC3

I have created a kit and ready made boards in 2021 and later, and there is a user group on Facebook too :

https://www.facebook.com/groups/z80mbc3

Recently, I have enhanced the IOS firmware with XON-XOFF flow control, and it also possible to select lower Baud rates on that board.
If you wish to experiment with this one ( it is still to be tested thoroughly ) then please reply or message me in the above group.

keirf, 2023-02-19:
Is this posted to the correct repository? This is the issue tracker for FlashFloppy floppy drive emulator.

eprive, 2023-02-19:
No, not subject of flashfloppy. It is a Z80-MBC2 subject.

But I noticed that you posted the issue on serial flow control there, and my answer is (currently) only intended for you who posted the question.

I have only access to the Z80-MBC3 group on Facebook, and the Github Z80-MBC3 repository.
It only has been released last week to the other group administrator (Carlo) for other testing purposes.

I don't want to publish it here on Github before it has been tested by an end user.

Willem ( eprive@gmx.com )

closed 2023-04-21 by keirf