keirf/flashfloppy #765: Flash Floppy in gotek installed in Ensoniq SQ80

dcarr3096, 2023-02-28:
Not sure ita the right forum but I'll ask...I was able to flash the gotek firmware, installed Flash Floppy successfully. Created about 20 floppies on it using the HxC program, the Ensoniq SQ80 lets me format each drive successfully and lets me access the disk load/copy/save page. I try to save a bank or a sequence and the SQ80 tells me it saved successfully but when I try to access it to load it back into the SQ80 it says the disk is blank. I've tried it on several consecutive disks to make sure it wasn't me selecting the wrong disk but all the disks say they're empty. Any help would be greatly appreciated!

keirf, 2023-02-28:
Did you format on the SQ80 or were the image files preformatted? What is the FlashFloppy firmware version? Can you zip and attach an image file after writing?

dcarr3096, 2023-03-05:
Well, I flashed the Gotek SFR1M44-U100 with the Artery programmer and then installed Flash Floppy 4.6a. I got the SQ80 with a bad drive and no floppies so I couldnt make an image from any existing files. Tried creating 100 blank image files with Batch Manage Tool 1.40V but the SQ80 only let me format 1 and whenever I tried to save a bank or Sequence it would only get to 78% and then give me an error message on the SQ80. Then I used The HxC Floppy Emulator and tried different formats until creating an 720 KB DSDD FAT12 MS-DOS image. Loaded 20 of those onto the thumb drive and the SQ80 recognizes them and lets me format it successfully on the keyboard. Once formatted I save either a bank or Sequence and it saves it successfully. When I go back to load it all the files on the screen appear blank and wont let me load anything.

dcarr3096, 2023-03-05:
I did create two DSHD 1.44mb disks, one I saved as an .hfe file and the other one as a .img file and when I try to format both they get to 79% when they stop and I get "Disk Format Failure, Disk is unusable. Pess any button to continue" on the keyboard screen. Maybe Im not making the images correctly or using the correct program to make them but Ive tried looking for an SQ80 image file online and havent been successful.

keirf, 2023-03-05:
Try latest v3.39 firmware instead of experimental v4.x

dcarr3096, 2023-03-05:
I updated the Gotek to Flash Floppy version 3.39 successfully. Now the SQ80 still sees the drive but says there is no disk in the drive no matter what file type or size I load. Doesnt matter if its .hfe or .img, 720kb or 1.44mb, it still says there's no disk in the drive. When I press the formatting button on the SQ80 the green light on the gotek lights up for a second so it seems its attempting to find the disk. Should I try other versions?

keirf, 2023-03-05:
Stick with 3.39. Have you tried different interface settings (eg. with and without JC jumper)?

dcarr3096, 2023-03-05:
Yes, I have the jumpers set to S0 and S1, that's how it was semi working before. If I remove S1 I get a Error 34 on the gotek. Should I try other jumper combos?

keirf, 2023-03-05:
Jumper setting will not cause Error 34. You will need one of S0 and S1 but not both. Try putting the spare jumper at JC instead.

Error 34 means no valid image file found on the usb drive by the way

dcarr3096, 2023-03-05:
Update:
Tried moving the spare jumper to the JC position and instantly blew two fuses in the SQ80s PSU lol Luckly I had a bag of spare fuses and its back up and running. Cant say for sure if it was due to the jumper or if it was just coincindence but it happened as soon as i turned it on so I dont think ill try that again. 
Regarding the error message youre right. I saw the Error 34 message after updating to version 3.39. Dont know if it has any bearing on this. If I remove the S1 jumper it just doesnt see the drive and the green light on the gotek does not turn on  when trying to format or send any other command from the SQ80. I'm going to try to duplicate what semi worked before to see if the keyboard sees a disk in the drive. Any other suggestions would be greatly appreciated!

Gkiokan, 2023-04-23:
Any news on this?  
We got the same problem so far that we have "no disk", Jumper set on S1 which worked with the original Gotek FW before.  
Now the disks seems to be loaded, as the display turns from 0 _ to 001 or 002 etc.  
But the Device (in our case KORG Trinity) does show "no disk".  

Tried with jc and imbpc as interface, host is set as akai but shouldn't make much difference.  

The usb is filled with the ff.cfg and the filled image files FIMG0001.img - FIMG0002.img.  

Any Ideas?

keirf, 2023-04-23:
KORG Trinity different host than Ensoniq SQ80. I would expect S1 + JC to work. Are the IMG files known good? What exact size are they, in bytes?

dcarr3096, 2023-04-23:
It would not work for me fully on the SQ -80. No matter what I did the SQ80
would say it saved it onto the Gotek, and the Gotek behaved like it was
saving the data, but then it wouldn't load the info saved. I was working on
an Emax 2 also so I repurposed the Gotek for that and it works fine without
having to reprogram it.

Gkiokan, 2023-04-23:
> KORG Trinity different host than Ensoniq SQ80. I would expect S1 + JC to work. Are the IMG files known good? What exact size are they, in bytes?

Thanks for taking your time. The Image files has been created with the Batch Manage Tool v1.40, then exported as IMG and put on are fresh formatted usb stick.  

I believe this is just a configuration issue. FYI: The reason we flashed from the native Gotek FW to FF was that we couldn't trigger the ready signal on disk change for the host. So Disk Changes wasn't recognized by the Host and this is a big trouble.  We can see, that FF does some reading which I assume to work as expected. 

We will try later to set S1 + JC Jumper and see what happens. 
What could cause the Host to see the FF as "no disk" besides the missing JC jumper?

Maybe if this gets more into deep it would be reasonable to create an own Issue to find a matching configuration.

keirf, 2023-04-23:
Getting disk detection working usually is a matter of correct configuration of signals such as 'disk-change' and 'ready' across floppy pins 2 and 34. That is what jumper JC, and the `interface` and `pin02` and `pin34` settings are all about.

closed 2023-06-27 by keirf