keirf/flashfloppy #768: MPC2KXL TYPE=NODISK

pyeatt99, 2023-03-11:
Hello
I got a Gotek recently and just updated the flash floppy software
I’m unsure of which disk images to use to my MPC will read it
When I update flash floppy and my MPC is still loading, I can see it, but then I go to load a file and it’s gone
Any workarounds?
[image]
[image]

keirf, 2023-03-11:
What are you trying to do? Format a blank HFE image file? Have you tried jumper S0 vs S1? Have you tried jumper at JC vs not?

keirf, 2023-03-11:
Unfortunately I have no prior reports for Akai MPC hosts, so you'll have to do some experimentation. If you look here for other Akai models https://github.com/keirf/flashfloppy/wiki/Host-Platforms#akai you can see that settings vary! Plus you then have to get working disk images (even if that just means formatting blank HFE images on the Akai host). Setting up for a new untested host system requires trial and error I'm afraid.

keirf, 2023-03-12:
There is a turnkey solution here: https://www.ebay.co.uk/itm/134071609375

So it does/can work. If you work it out, let us know and I can update the wiki with your steps!

closed 2023-03-12 by keirf