keirf/flashfloppy #769: Technics KN2000 support

Erriez, 2023-03-12:
I've a full functional Technics KN2000 keyboard from 1993 and I'm interested in this floppy emulator hardware/software project. The original built-in floppy drive supports 720kB floppy's only and looks like a propitiatory filesystem format.

Here is the disk controller schematic with pin-out:
[image]

Any suggestion if this will work with a `Gotek Floppy Emulator met Rotary Encoder` or other emulator and how to proceed?

keirf, 2023-03-12:
First you'll need an adapter from 34 pin IDC on the Gotek to the 24 pin probably FFC here. Likely exists as I think some other musical instruments use this 24 pin interface. After that you will need disk images too.

Erriez, 2023-03-12:
I received an answer from gotek.nl: The pin-out is different as you confirmed (thanks @keirf). The Gotek part should not be a problem, but the challenge is the dimensions to physically fit into the keyboard. I've added related links in the description of this issue.

keirf, 2023-03-12:
So it looks like gotek.nl can sort you out, from the link you added?

Erriez, 2023-03-12:
Not really, because they don't have a KN2000 device to test it. So I'm at my own and is a risk as I currently have a working floppy-drive.

keirf, 2023-03-12:
Do they have the physical adapter available?

Erriez, 2023-03-12:
A physical adapter is not available due to the low number of KN2000 requests.

keirf, 2023-03-12:
I see. Without such an adapter you are stuck and Gotek cannot be used with this keyboard.

Erriez, 2023-03-12:
Looking for this adapter:
[image]

keirf, 2023-03-12:
There's a good UK seller who sells a "plug and play" package including Gotek, adapter, and USB drive. The current listing for KN 2000: https://www.ebay.co.uk/itm/134022605591

Reasonable prices for "turnkey", and very friendly, so if you want any pre-sales advice I suggest dropping them a line.

closed 2023-03-12 by keirf