keirf/flashfloppy #771: Akai s950 + gotek flashfloppy v3.30

boombipbass, 2023-03-18:
hi,

I am really confused, I used different subs now(3, all work still normally on my pc are fat32 formatted et all)

The usb light up when in my pc or MacBook, they don't in the gotek. I tried updating the firmware(the latest stable one) via flashfloppy .upd (and the other to be sure)

hold the buttons, started the machine, let the buttons go, screen keeps hanging in ff firmware update [usb]

I recall it did work one time when I initially installed the usb, meaning that I could see the .hfe files on the gotek LCD but when trying to load them via "disk" I got the oops error.

Some moments after this issue I eventual;y could go to disk menu but I wasn't able to see the .hfe files anymore on the gotek LCD.

I am now 2 months further, I might have pressed some buttons accidentally as a desperate move hence I messed it up I don't know?

But I do know it just doesn't work, and I am baffled. I also factory reset it but nothing, just nothing :-(

boombipbass, 2023-03-18:
update:

finally after 1 month absence, I noted my usb blue light giving small led burst, hence meaning that it is connected.
I tried the flashfloppy update as said here: https://github.com/keirf/FlashFloppy/wiki/Downloads

But it is still at v3.30

nonetheless I added a ff.cfg (s950 oled native) and some .hfe but it doesn't show up on the gotek itself but I am capable to go to disk and then scroll there, what should I do?

boombipbass, 2023-03-18:
argh it says disk is unprepared in the s950 menu, also I noted it said on the right uppercase the s900 text, could this be a mistake from me?

keirf, 2023-03-18:
If the firmware update is stuck at [usb] then the Gotek doesn't like your usb stick. If it's an old version of the bootloader (version is displayed before you press a button and the display changes to [usb]) then perhaps the problem is that your stick is formatted gpt. It might need to be formatted mbr. On windows the Rufus tool can do this.

boombipbass, 2023-03-19:
> If the firmware update is stuck at [usb] then the Gotek doesn't like your usb stick. If it's an old version of the bootloader (version is displayed before you press a button and the display changes [...]

just downloaded Rufus thanks, will try it again, what boot selection should I choose?

keirf, 2023-03-19:
Boot selection should be "Non bootable". This way Rufus should allow you to format the USB drive without specifying an image to write to the formatted USB drive. This is what you want: Format only.

boombipbass, 2023-03-24:
Hi I am sorry I thought I replied, it worked with the usb stick but the usb stick the 1 gb still isn't seen.
I will now try the 128 gb one that couldn't update but was able to go to the disk menu(albeit not being able to format there or save or load anything or show any of the .hfe files(not on the disk menu nor the OLED screen despite having the .cf file and some blank disk title files.hfe on there)

boombipbass, 2023-03-24:
I did notice after I formated the 1 gb disk there were two files on the usb9might be autoboot files? since the folder showed up right away)

boombipbass, 2023-05-03:
any clue what is wrong?

keirf, 2023-05-03:
So.. you have a USB stick where when you insert it the FlashFloppy display now updates to show that an image is mounted? Shows track display etc?

boombipbass, 2023-05-14:
well I cam show the menu from the old and I can put it on the usb but when I press the turning button to use it it doesn't work out

closed 2023-06-27 by keirf

boombipbass, 2023-07-02:
Why you closed the topic since the solution itsnt even solved :(

Closed #771<https://github.com/keirf/flashfloppy/issues/771> as completed.

keirf, 2023-07-02:
If I keep this ticket open for another couple of months do you think there would be progress? What do you expect to happen next? Your support questions would be better asked in the Discussion area on GitHub, or in the FlashFloppy Facebook group, as this Issues area is really intended for firmware issues (bugs).

keirf, 2023-07-02:
If you like I can convert this ticket into a Discussion, unless you want open a fresh Discussion summarising your problem?