keirf/flashfloppy #774: Gotek Flash Floppy v3.39 MIDAS OLED Display

Aether2013, 2023-03-24:
Hi all,

Recently added a Midas MCOB21605G1V OLED display to my Gotek in my A1200, the Gotek works fine in terms of loading games and displaying game names etc but I am unable to control the screen functions and the screen won't switch off and go into standby mode when no activity detected. I have tried with the original smaller OLED display that comes with the Gotek and it works fine.

Any idea, seems to be a compatibility issue with the Midas OLED and the Gotek?

Link to the OLED display

https://au.element14.com/midas/mcob21605g1v-ebp/display-oled-cob-16x2-blue-on/dp/2769684

keirf, 2023-03-24:
It's a character display not a pixel-addressable display like the usual OLEDs used with Gotek.

FlashFloppy sees this as an LCD display and turns it off by turning off the backlight. But there is no backlight. I suppose we could have a config option to blank the LCD. But very very few users have this display type.

Aether2013, 2023-03-24:
Thanks keirf, makes sense.  Would you know who could assist with the Config option to blank the screen, if that's possible?  I have modified the A1200 case to fit the OLED screen and kinda stuck with the size now 😬. Can't find a pixel addressable OLED with the same dimensions as this one to fit to the case.

keirf, 2023-03-24:
Let me have a think. Can't be very hard.

Aether2013, 2023-03-24:
Thanks, would really appreciate that 👍

Aether2013, 2023-03-24:
Absolute legend, thanks heaps that firmware does exactly what I was looking for!  Appreciate you assisting with such a unique request and the quick replies and firmware mod, can't thank you enough champ 👍👍👍

keirf, 2023-03-24:
No problem. It seems logical to enable this all the time for any LCD display. So it will be in the next official release, but that could be a while (months?) away. But there is no reason you can't continue to use this intermediate firmware for now, until the next numbered release.