keirf/flashfloppy #776: Yamaha QY700 Gotek crash in Executing... (no Save/Load)

daftpunker, 2023-04-04:
Hi all

After installing my Gotek SFRKC30AT 3 3.35 – 32KB on the Yamaha QY700 jumper on S1 and JC I can't save or load, In attempts the qy700 remains planted in "Executing" and the gotek does not react

[image]

Even trying to load not the qy700 does not see anything to load even though I select them from the gotek.

I enclose the tested USB sticks:

[Qy700 Trial1.zip](https://github.com/keirf/flashfloppy/files/11145986/Qy700.Trial1.zip)

[Trial2.zip](https://github.com/keirf/flashfloppy/files/11145992/Trial2.zip)

Obviously I also tried to do the factory reset but nothing to do.
Do you have a usb stick ready and working to attach to see if it's a problem with the ff.cfg file?

Thank you for your attention

daftpunker, 2023-06-02:
No one showed up... can anyone help me? I also tried to format on QY700 the file but .hfe nothing to do whether it is a save or a format or a load it hangs in the initial phase and remains in Executing

daftpunker, 2023-06-02:
Format
[image]
[image]

daftpunker, 2023-06-02:
Save
[image]
[image]

keirf, 2023-06-02:
In the first photo back in April the Gotek has a FAT02 error, which is an internal error in the FAT library used by FlashFloppy. Subsequent photos don't show this error though.

keirf, 2023-06-02:
These disk images are regular PC 1.44MB. You should instead be using IMG files rather than HFE. You will not be able to reliably format (or possibly even save to) HD HFE image files from a Gotek with 32kB RAM. IMG files are much easier to write to.

daftpunker, 2023-06-02:
Hi Keirf, thanks for the quick response.
1 I changed the jumper configuration, now the jumper is only on S1.

2 I factory reset the Gotek via the menu

3 I have the files that I took over from the first new design QY700. I put them on the USB stick, just the CFG file and the image file. I will send you the contents of the test stick.
Returns error 31
[QY700.zip](https://github.com/keirf/flashfloppy/files/11638687/QY700.zip)
[image]
[image]
[image]
[image]

keirf, 2023-06-02:
This is because the IMG file is somehow 512 bytes smaller than it should be. It needs to be 1474560 bytes long. Perhaps the HFE image file contained a corrupted sector?

Get yourself a good 1474560 byte IMG file and you should get further.

keirf, 2023-06-02:
Here is my conversion of qy700_Demo2.hfe to IMG format. Perhaps this is useful?
[qy700_Demo2.zip](https://github.com/keirf/flashfloppy/files/11638947/qy700_Demo2.zip)

daftpunker, 2023-06-03:
Hi, I tested but nothing changed.
I enclose stick and photos
[QY700keirf.zip](https://github.com/keirf/flashfloppy/files/11641968/QY700keirf.zip)

Test Load: 

[image]
[image]

Test format: 

[image]
[image]

daftpunker, 2023-06-03:
Always the same result even with this test

[QY700 2.zip](https://github.com/keirf/flashfloppy/files/11642000/QY700.2.zip)

[image]

[image]

[image]

[image]

[image]

[image]

keirf, 2023-06-03:
Looks fine on Gotek side. Usb stick? No more ideas.

closed 2023-06-25 by keirf

daftpunker, 2023-06-25:
I tried with different keys formatted and read by other goteks but nothing to do. It remains at T:00.0