keirf/flashfloppy #777: Emu ESI-32 Gotek KO SAVE error “Floppy FIFO Overrun”

daftpunker, 2023-04-04:
Hi guys, I'm having problems saving with my gotek SFRKC30AT 2 on my sampler Emu ESI-32

I enclose a zip with all the contents of the trial usb stick:

[EMU Esi32.zip](https://github.com/keirf/flashfloppy/files/11146122/EMU.Esi32.zip)

The problem occurs only in the save operation, after a few seconds 
[image]

the second execution bar stops with this error: "Floppy FIFO Overrun"

[image]

The problem occurs only on the save procedure. Gotek instead reacts well in loading operations

keirf, 2023-04-24:
Maybe the USB drive is too slow and you have a 415 based Gotek (does it say 32kB on the title page when no USB stick is inserted?). Try another stick. Also consider getting a AT32F435 based Gotek now they are more readily available. Much bigger buffers (384kB RAM).

daftpunker, 2023-05-23:
[image]

keirf, 2023-05-23:
You have no chance of reliably saving to a high-density HFE image from a 32kB Gotek. You either need a 435-based Gotek, or here you switch to 1440kB IMG files. You can convert the HFE images to IMG using HxC software, or Greaseweazle tools. Unlike EMU II (and other EMU double-density instruments) the EMU HD sector format is same as IBM PC.

keirf, 2023-05-26:
Any more to say on this one, or shall I close it?

daftpunker, 2023-05-28:
Hi Keirf, changing usb stick with a new one works! He deceived me because the old key works with all the other samplers I have but on this Emu it is indigestible! Yes you can close the post, thanks

closed 2023-05-28 by keirf