keirf/flashfloppy #785: IBM RS/6000 43P 7248-100

blackbit42, 2023-04-30:
My main question is:
> How does one configure the equivalent of jumper setting `S1` in `FF.CFG`?

I have a `IBM RS/6000 43P 7248-100` that I want to transition from classic floppy disk drive to a Gotek emulator.

The floppy drive is an `Alps DF334H012A` which does not appear to have any jumpers or dip switches. It's a 3.5" drive with a regular 34pin ribbon cable.

My main usecase for floppy/Gotek is to load the so called `Systems Management Services` (SMS) disk. That is an operation that is initiated from the boot firmware, i.e. not from a running operating system. That floppy seems to be a regular FAT12 formatted disk. I believe there is no special boot sector magic happening, as the 'disk creation utility' from IBM is just a self-extracting zip file. Instructions say to run that on a formatted disk and you are done. My assumption is that the firmware searches for a file with a particular filename on the disk.

I have connected a Gotek PCB to the machine that has an `Artery AT32F415` uC. An img image of that disk is on the USB drive that I use. The firmware is not able to read the emulated disk.

I've experimentally determined that having the PCB jumper set to `S1` leads to a blink of the activity LED when the firmware tries to load the disk. Other jumper settings leave the LED still.

Unfortunately I do not have a 7segment display or an OLED that I could hook up to the Gotek, so I don't get a lot of information out of the device. Therefore I've loaded the logfile variant of FF, version `3.39`, so that I can get some information about what is happening during unsuccessful access of the emulated floppy. In order to be able to flush the logfile to the USB drive, I've created a `FF.CFG` with `twobutton-action = eject`, though then I ran into an issue. When a `FF.CFG` file exists, the settings for `interface` and `host` take precedence over the jumper setting IIUC, though I wasn't able to figure out, based on [the wiki page for FF.CFG](https://github.com/keirf/flashfloppy/wiki/FF.CFG-Configuration-File), how I configure the equivalent of jumper setting `S1` in `FF.CFG`. Hence, the question I've asked above.

keirf, 2023-04-30:
You probably need jumpers at S1 and JC

blackbit42, 2023-04-30:
That worked. Thanks!

This is the logfile that I am getting:
```
** FlashFloppy 3.39
** Keir Fraser <keir.xen@gmail.com>
** github:keirf/flashfloppy

Build: Mar  1 2023 12:34:30
Board: Standard
Config: Flash Slot 1 (ver 2, size 74)
Interface: IBM PC (pin2=high, pin34=chg)
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
I2C: Invalid bus SCL=0 SDA=0
Display: 2-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 1307
 PID : 0163
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : P Technology
 Product : USB Mass Storage Device
 Serial Number : 00000000000DDC
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 42 items
0:F: 'sms.img' (EJ)
Cache 42 items
Mode: Native
Current slot: 0/0
Name: 'sms' Type: img
Attr: 20 Clus: 0000016b Size: 1474560
Before: After: Before: After: Current slot: 0/0
Name: 'sms' Type: img
Attr: 20 Clus: 0000016b Size: 1474560
```

Can anything relevant be deduced from that?

keirf, 2023-04-30:
For interface and drive select settings unfortunately not really. You have to know or know what to try.

blackbit42, 2023-04-30:
This log was written with the `JC` and `S1` jumpers closed and I got the 'blink' of the activity LED when the firmware tried to access the disk.

Is there an expectation to get an indication of that access attempt in the log?

keirf, 2023-04-30:
No because drive select is not logged.

closed 2023-05-05 by blackbit42