keirf/flashfloppy #786: GD32F105RBT6 Support

M1kerochip, 2023-05-03:
I bought a GD32 device to try out, last year, from LCSC, when there was a shortage of STM32 ics, and completely forgot I bought it.

Do you think it needs specific support added, and would there be much work in supporting this device?

https://www.gigadevice.com/product/mcu/arm-cortex-m3/gd32f105r8t6

It says it's fully compatible with an STM32, but, I'm not having much luck!

I installed it on an openflops pcb, and tried to flash it, and that's where the fun began!
The official flash tools crash when selecting a .hex file to flash in windows.

So, I used the ST tool DfuFileMgr to convert the .hex into two bin files.
I erased the GD32, full erase, then programmed each of the bin files, 0x08000000 and 0x08008000 erasing only the necessary pages, and wrote both bin files to the GD32, at the memory locations of the respective file, 0x08000000, and 0x08008000.

I also downloaded the flash contents after I wrote them, to a .bin file, attached, if you want to take a look.

It doesn't attempt to start FF though, when I power up the OpenFlops.
I can see it's getting power, but, it doesn't do anything!

I presume my soldering is ok, since I can flash the firmware in both dfu mode, and with a uart adapter.
I also soldered up another OpenFlops pcb, and used a genuine STM32, and it works perfectly.

(Not that the GD32 isn't genuine, it is, it's just not from ST obviously)

[upload.zip](https://github.com/keirf/flashfloppy/files/11380612/upload.zip)

M1kerochip, 2023-05-03:
Sorry, typo on my part.

It is indeed RBT6. (It exists!)
[image]

https://www.gigadevice.com/product/mcu/arm-cortex-m3/gd32f105rbt6

M1kerochip, 2023-05-03:
C81551 is the LSCS part number.

https://www.lcsc.com/product-detail/Microcontroller-Units-MCUs-MPUs-SOCs_GigaDevice-Semicon-Beijing-GD32F105RBT6_C81551.html

keirf, 2023-05-03:
One thing to try is flashing the blinky firmware that ships with greaseweazle firmware and connect over serial at 115200 baud to see what diagnostics you receive there. You can also run debug version of FlashFloppy to see whether you get  any serial output at all (that's at 3 mega baud).

keirf, 2023-05-03:
More generally I more eagerly support the AT32F435 now so support for 105 clones is not something I'd put much effort into.

M1kerochip, 2023-05-03:
Ah, ok! I'll give the blinky a go. I never thought of the debug version, too.

Are any of the ST32F435 a drop in replacement for the STM32F105?

keirf, 2023-05-03:
It is a drop in replacement for STM32F105! But not for AT32F415 (which has fewer power pins) so the Gotek factory 415-specific boards aren't compatible. But the original STM32F105 boards, and clones of that design, are!

M1kerochip, 2023-05-03:
Ah, so, I presume then, the LQFP-64 AT32F435RGT7 should just work on the OpenFlops from SukkoPera.

Looking at the spec sheet, R is 64 pin, M is 4M, G 1M, and C 256K, and T is lqfp.
So, AT32F435R(M/G/C)T7 should be fine?

Yeah, strange they're not compatible. Well. Pin compatible.

keirf, 2023-05-03:
Yes exactly so.

M1kerochip, 2023-05-03:
So, is the F435 then, almost fast enough to drive a small colour display, were we could display a bitmap/png of the matching disk image name, to a small screen?

So, diskA.img gets loaded into flashfloppy, but diskA.png gets displayed on a tiny screen.

That's what I'd love to see! A tiny picture of my floppy 😄

keirf, 2023-05-03:
Oh it could definitely do that.

M1kerochip, 2023-05-03:
Well, if you ever think about adding it as a feature, and need anyone to beta test that, let me know 😄 

*looks up SPI displays* Looks like a nice SPI 240x240 IPS panel costs more than a Gotek...

M1kerochip, 2023-05-03:
Actually, Pimoroni have a nice looking tiny one: https://shop.pimoroni.com/products/1-3-spi-colour-lcd-240x240-breakout

keirf, 2023-05-03:
Easy to find for very cheap on ebay

M1kerochip, 2023-05-03:
OK, I'm going to close this for the moment. It'll be the weekend before I try the blinky/serial out, and, with an artery F435 costing £3 each, I'll just get some of those, and swap it out I think. Least amount of work, all-round.

closed 2023-05-03 by M1kerochip