keirf/flashfloppy #789: Sharing usb drive between Gotek with different configurations: possibility of tying FF.CFG to specific Gotek?

Yrouel, 2023-05-10:
I have two Gotek connected to two PCs with different configurations, one has a 0.96" OLED (128x64) and the other the normal 7-segment display.
If I use "display-type = auto" to make FF.CFG work for both, the OLED is configured with the default look which means having for example the first line half yellow and half blue and if I use "display-type = oled-128x64" the 7-segment display doesn't work.

Would it be possible to tie a configuration to a specific Gotek somehow? Do Gotek have for example a unique id (like the serial of the chip) you can tap?
The config file could become FF.ID.CFG in that case

keirf, 2023-05-26:
FF.CFG entries get cached in the Gotek. So simply introduce an FF.CFG to each Gotek with the appropriate display-type line. Then delete/comment the display-type line. Job done!

keirf, 2023-05-26:
Another way of managing it: You can even have two FF.CFG files on your drive, FF.Gotek1.CFG & FF.Gotek2.CFG (or somesuch). Then if you ever need to reinstantiate the config on one of your Goteks (eg cached config gets reset for some reason) you can copy the FF.GotekX.CFG file to FF.CFG temporarily.

Yrouel, 2023-05-26:
Thanks for the proposed solutions, I think it's good enough.
I thought the configuration would've been reset changing back and forth

closed 2023-05-26 by Yrouel

keirf, 2023-05-26:
As long as the Gotek doesn't see an updated version of a particular config item in a later FF.CFG, it will remember what you set.