keirf/flashfloppy #792: Help on installing the Rotary Encoder?

iosoft, 2023-05-22:
Hello,

I want to install _Rotary Encoder_ (KC30 Rotary Header) with my **SFRKC30.AT2** Model.
I have two Rotary Encoders. One has an extra resister -
[image]

Which one to use/install?
Pls suggest.

keirf, 2023-05-22:
Try them both? It's hard to know without knowing that specific rotary board design. They may well both work.

iosoft, 2023-05-22:
> Try them both? It's hard to know without knowing that specific rotary board design. They may well both work.

They both are exactly same looking, just R1 (a Pull-Up Resistor) is missing in one of them.

keirf, 2023-05-22:
FlashFloppy doesn't need pull-ups so either will work in all likelihood.