keirf/flashfloppy #798: ADF Filename Change

shan2752, 2023-06-18:
Hello,

Thanks for all the great work in creating Flash Floppy!  Is it anyway possible for an ADF file name to be changed within Amiga OS and the change reflected on the USB/SD image file?

This would be extremely helpful if it were possible.  Thanks!

keirf, 2023-06-18:
Unfortunately not, since the Amiga host knows nothing about the image file names. All it can change is the amigados volume label within the image file itself.

shan2752, 2023-06-19:
Is there a way to rename an adf file within Flash Floppy?  For example, I have a blank disk ADF called Blank01.adf, I dms extract an archive to this blank, which I want to rename.

Thanks!

keirf, 2023-06-20:
Nope I'm afraid not. It may get added in future.

closed 2023-06-20 by keirf