keirf/flashfloppy #800: Need Help: using FlashFloppy as 720KB Drive for PC-XT system

iosoft, 2023-06-19:
Friends,

I installed FlashFloppy on my Gotek (SFRKC30.AT2).
But I have a 2GB Pen-drive only. This is the lowest size I have.

Now -
- How to configure it to work as 720KB Drive for PC-XT system?
- How to create and upload Disk image(s)? One DOS Boot image.

Regards

closed 2023-06-20 by keirf

iosoft, 2023-06-20:
> Format stick FAT32 and place 720kb image files on the stick. Place jumpers at rear of Gotek at S1 and JC. Gotek will automatically detect img files and let you select between them.

Thank you for your kind time and suggestion.

1. So, I need to set 2 Jumpers, S1 and JC (currently I have only S1)

[image]

2. Where can I find/create these 720kb image files?
Specially a DOS Boot image.

P.S. Doing it 1st time, so.....

keirf, 2023-06-20:
Finding images is part of setting up on a new system. For example I can google around for PC DOS boot disks and find this link for early PC-DOS, which looks promising (160kB disk image): https://winworldpc.com/product/pc-dos/1x

I'm sure MS DOS install media images must be out there somewhere too...

iosoft, 2023-06-23:
Friend, one more small question.

My PC-XT (clone) supports only 360KB or 720KB Floppy Drivers, selected by Jumpers.
So, I set the jumper as 720KB and also the FlashFloppy as 720KB as you suggested above.

[image]

**Question:** 
So I have to upload only 720KB size .img files?
I cannot use/upload any 1.44MB images, right?

omikron88, 2023-06-23:
1.44MB Floppy uses double bitrate.

iosoft, 2023-06-23:
> 1.44MB Floppy uses double bitrate.

So, not possible right?

omikron88, 2023-06-23:
No, if the controller cannot be switched for that.

iosoft, 2023-06-23:
> No, if the controller cannot be switched for that.

Ok. I was just asking.... you know that XT-IDE thing can make old system to use High-capacity drives in old systems..... may be that is also possible for FlashFloppy or combined with XT-IDE+FlashFloppy.

That would be cool!