keirf/flashfloppy #802: XDF (OS/2 format) write problem

Zyntax3rror, 2023-06-24:
When using xdfcopy in DOS/Linux some data gets corrupted when written using Gotek. It works fine on a real floppy drive => 1:1 copy)

I used 384KB Gotek + ff 3.38

DOS: **xdfcopy test.dsk A:** 
Linux (mint 19.3): **sudo xdfcopy test.dsk /dev/fd0**

In my image the data gets' bad @ 0x2000- 0x2BFF in the first file and is filled with a pattern.
Different pattern in Linux but on the same location.

keirf, 2023-06-24:
Can you attach before and after dsk image files for inspection?

keirf, 2023-06-24:
That is the source dsk, the destination dsk before copy, and the destination dsk after copy. Zip them all up and attach.

Zyntax3rror, 2023-06-24:
yes, no problem .I will re-do the test and post later

Zyntax3rror, 2023-06-24:
File uploaded on first post

keirf, 2023-06-24:
What does the destination dsk image look like *before* the copy?

Zyntax3rror, 2023-06-24:
For this test I used the same source.dsk as destination.

keirf, 2023-06-24:
Ok, thanks. I can see that three of four sectors on head 1 of every cylinder >= 1 is corrupt. These areas contain bogus 256-byte sectors. This is probably from the format step, and then does not get overwritten by real data from the actual copy, for some reason.

Could you try writing on Linux with the `-n` option, copying to a fresh destination.dsk (ie. a fresh copy of your source.dsk) and attach that? It would make sense that, if writes are missed, this should be entirely or almost entirely okay, right?

keirf, 2023-06-24:
Actually the format step is probably bogus, as I think there is deliberate mismatch between sector headers and sector data sizes, which will no doubt confuse FlashFloppy. So the `-n` xdfcopy case is probably most interesting.

Zyntax3rror, 2023-06-24:
I tested three times and it was a 1:1 copy every time!  So no point uploading it since it was the same as the source :)

Zyntax3rror, 2023-06-24:
I tested to create a blank 'xdf' using

sudo xdfcopy -D A /dev/fd0 -1

I used blank.dsk as destionation this time and the same source as before

[BLANK.zip](https://github.com/keirf/flashfloppy/files/11857469/BLANK.zip)

keirf, 2023-06-24:
Okay the differences on cylinder 0 appear to be simply because those areas aren't represented in an XDF image file. So xdfcopy doesn't copy them, unsurprisingly.

On cylinders > 0, it seems that nothing on side 1 gets written at all. That has to be a bug somewhere I guess. I wonder what FlashFloppy is doing wrong...

keirf, 2023-06-24:
Best guess is that the special skew I apply to side 1 on XDF images and nowhere else is not being handled properly on writes. That would not be surprising. I will see if I can confirm that somehow...

keirf, 2023-06-24:
Okay, I see the bug. I have a fix but unfortunately it is hard for me to directly test it. There is an automated build available at the link below. You will need to be logged in to GitHub to see the Artifact download link at the bottom of the page. You should download it, unzip it and then the *non* debug zip contained within, and then update firmware as usual. You can afterwards update back to a proper release version of the firmware if you want to.

https://github.com/keirf/flashfloppy/actions/runs/5365322741

Zyntax3rror, 2023-06-24:
Here is the test using new firmware and -n in linux 

different at address 2800 (all 00)
[NLINUX_NF03.zip](https://github.com/keirf/flashfloppy/files/11857805/NLINUX_NF03.zip)

I will test again

keirf, 2023-06-24:
Differences in the following ranges in the DSK images do not matter: 2800-2E00, 4A00-5400

These ranges are not part of a physical XDF disk, and are not included in the xdfcopy copying operation. I'm not actually sure what they are for. If you can't see any differences outside those ranges, I think we are all good.

Zyntax3rror, 2023-06-24:
ok, I also tested xdfcopy in DOS and image file got same crc as nlinux one.
I extracted the files and compared to source and all files crc correct(DOS+nlinux) = all files contains  correct data :) 

I will do some more test's later

Zyntax3rror, 2023-06-24:
Disk Copy Fast (DOS) also works now. I exctracted the files and all crc correct vs source

Thanks for fixing this.

I think you can close this one ;)

keirf, 2023-06-24:
Cheers, I will commit the fix to master and this will automatically close the ticket.

commit 2935de98 (2023-06-25): xdf: Fix writes to side 1 (with bitcell-granularity track skew).

Fixes #802
closed 2023-06-25 by keirf