keirf/flashfloppy #803: IBM PS/1 write issue with stable branch

theruler, 2023-06-24:
HI, I have a model 2011 286 IBM PS/1, in 2018 I've replaced the old broken original floppy with a Gotek (cpu 72 ram 64) and recently discovered that I cannot write on images a file bigger than 6 KB. I've probably always performed the opposite only, never saved to disk.
Tried all FF firmware from 3.39 down to 3.28 with and without CFG files
Read and format NO PROBLEM at all. Write a 0.5-6KB file No problem, Writing a 10KB file (or more) ends up in INT 24 Fail, copy error.

NO ISSUE on the experimental branch 4.0a-4.6a.

keirf, 2023-06-24:
Probably fussy about write timing. The experimental branch can be tighter on this. Sometimes on the stable branch a different usb stick can help. What stick(s) have you tested?

theruler, 2023-06-24:
Actually never thought about USB speed. I've used only one, an OLD EMTEC 1GB fat32 formatted. Tomorrow I will try a newer one.
Is there a parameter I should change in FF or IMG to trim the timing?
Cheers, thanks for the prompt reply. I LOVE your work.

theruler, 2023-06-25:
Ok, tried two other sticks with 3.39 and the problem is solved. THANKS!

keirf, 2023-06-25:
Great! The I/O timing on 3.x is fundamentally different than 4.x

3.x writes back each written sector synchronously back to USB stick. The "read" of the track is paused while this happens. If a stick executes some writes really slowly, the host can time out while this occurs.

4.x attempts to hide writeback latencies by deferring the writes to USB until: (a) All data has been read for the current track into RAM (then data can be written back while servicing the track read from RAM buffers); or (b) The heads are stepped to another cylinder, which provides a convenient latency cover behind which to hide the writeback.

Most hosts don't actually care about the 4.x improvements as long as the stick is fast enough.

closed 2023-06-25 by keirf