keirf/flashfloppy #807: The VFD Futaba display.

MikhaelKaa, 2023-07-07:
Is it possible to add support for VFD displays like futaba?
They have a uart 8n1 9600 interface. The source code is attached.
I tried it myself, but I had difficulties assembling the project.

[image]
[image]
[image]

#ifndef VFD_DRIVER
#define VFD_DRIVER
#include "main.h"
#include "usart.h"
#include "gpio.h"
#include <stdio.h>
#include <stdarg.h>
// dimming
#define DIM     0x04
// diaplay position
#define DP      0x10
// cursor off
#define CUR_OFF 0x14
// cursor on
#define CUR_ON  0x13
// reset
#define RST     0x1F
// bright
#define BR_0    0x00
#define BR_20   0x20
#define BR_40   0x40
#define BR_60   0x60
#define BR_100  0xFF
void vfd_init();
void vfd_print(const char *format ,...);
#endif /* VFD_DRIVER  */

// vfd driver

#include "vfd.h"

char vfd_buf[40];

void vfd_put_char(char c) {
    HAL_UART_Transmit_DMA(&huart1, (uint8_t*)&c, 1);
    HAL_Delay(1);
}

void vfd_init() {
    vfd_put_char(RST);
    vfd_put_char(DIM);
    vfd_put_char(BR_100);
    vfd_put_char(CUR_OFF);
}

void vfd_print(const char *format ,...) {
    int total_bytes;

    va_list argptr;
    va_start(argptr, format);
        total_bytes = vsprintf(vfd_buf, format, argptr);
    va_end(argptr);
    HAL_UART_Transmit_DMA(&huart1, vfd_buf, total_bytes);
}

keirf, 2023-07-14:
Quite different than the existing supported I2C interface. You want to start by getting debug logging on the primary USART1?

MikhaelKaa, 2023-07-14:
It differs very much from i2c.
Yes, I couldn't turn on printf, although I did the uart clocking and settings.
If it's not difficult, I would be satisfied with a project in which the output to uart works at a speed of 9600, another type of display (vfd) is added - I would try to write the output code to this screen, tested and made a pullrequest.

keirf, 2023-07-14:
It is automatically built and you can find the debug logging version under `out/stm321f105/debug/floppy/target.hex`

closed 2024-01-11 by keirf