keirf/flashfloppy #808: Buttons not working with LCD 20x04

robxbl69, 2023-07-10:
I already have several GoTek's with FF installed. Have already used various OLED's and Flash Floppy OSD, so to make this new one a bit different I thought I'd used an 20x04 LCD display.

The GoTek is a SFRKC30AT3 (LQFP48) with FF 3.40. Default config file with JC jumper installed. Works great with original 2 segment LED display and with 16x02 LCD (either using display-type = auto or display-type = lcd-16x02).

The 20x04 LCD display also works with auto or 16x02 specified (with short and missing lines) and if I specify display-type = lcd-20x04 the GoTek boots with the text on the two middle lines and then gives me 4 lines of text, but I cannot then change disk image. I don't have a rotary encoder connected, so cannot check if it's just the buttons not working.

If I pull the USB stick the screen reverts to the boot screen. Interestingly if I specify 20x03 I get 3 lines of working text and buttons. 19x04 also stops the buttons working.

I've tried changing the display order to something like display-order = 3,0,7,1 and it still locks up.

For now I'm using:
display-type = lcd-20x03
display-order = 3,0,1,7
display-scroll-rate = 200 (80ms is too fast on a LCD)

I'm not sure what else to try...

Thanks (in advance), Rob

keirf, 2023-07-11:
It sounds a bit like a bug! I have one of these displays I'm pretty sure, so I will have to try hooking it up to an AT3 model.

robxbl69, 2023-07-11:
If you don't have one, then I have spare and could send you one (I'm in UK also).

keirf, 2023-07-11:
I've found mine. I'll give it a whirl. The LCD I2C code is a bit different for 415 chips vs the old 105 chips. There's probably a buffer overflow with the larger LCD display data, or something like that.

keirf, 2023-07-12:
Actually the 415 isn't different; that's the newer 435 chip. I'm not sure what the problem here could be. I will try to repro though. I'm trying to think if it could be your specific display getting stuck. But it does update when you pull the USB stick so that seems a bit unlikely.

robxbl69, 2023-07-12:
I'll have another play later, but I'm wondering if it only happens with long (scrolling) file names.

If you want me to eliminate (or prove fault of) the LCD display (or I2C controller) I could hook it up to an Arduino and give it a hammering?

keirf, 2023-07-12:
Unlikely due to long filenames. Try a stick with just a short filename on it? Don't bother hammering it -- I should at least try to repro first .

robxbl69, 2023-07-12:
Just tried with filenames < 20 chars and still locks up immediately on 4 x 20. Having played some more it also locks up on 3 x 20 on occasions. So far no issues with 16 x 02.

I'll hook up a 1.3" OLED later this evening just to prove it's not a dodgy GoTek.

keirf, 2023-07-13:
I can reproduce the exact same issue. So. It's a bug.

keirf, 2023-07-13:
Actually, I can *sometimes*. It seems after power on, for a while. If I leave it a bit, the buttons start working.

robxbl69, 2023-07-13:
Okay, pleased you can replicate the issue.

Later I will try leaving it on for a while to see if my results are the same, but have tried previously pulling an re-inserting without any joy.

Also yesterday when powering on I just got gibberish on the display (2 lines of inversed random symbols I think). Only way I could fix this was to power cycle the Amiga (not just reset) and therefore power cycle the GoTek. 

This may have just been another random, but one time when I had the filename as the 4th row it behaved better (didn't lock up straight away, but did in the end).

keirf, 2023-07-13:
So this was due to a horrendous subtle bug in the timer code, and a recent speed bump for AT32F415-based Goteks to 144MHz (previously 72MHz).

It should now be fixed. Please can you try the automated build here (you will need to be logged in to GitHub to see the Artifact link at the bottom of the page, and you should unzip the non-debug version and use the update file as usual):
https://github.com/keirf/flashfloppy/actions/runs/5546119876

robxbl69, 2023-07-13:
I've tried the new version and it now seems to have stopped crashing/locking up :) 

However I have a new issue (sorry)... 

There's now an extra space down the left side and the single character overflow from line 2 is at the start of line 4 (for example the number "2" in the image below). Same happens when scrolling, the next character to appear at the end of line 2 appears on line 4 first :-/

[image]

I'd normally rename files so they don't have to scroll as I just don't like it, but will mention that even with the new update I notice the scrolling speed it a bit erratic. Sometimes steady, then it slows and then suddenly races away to catch up. Could this be the I2C controller on the back of the LCD rather than FF?

keirf, 2023-07-13:
I have a setup now so I can try to repro this too. Has it always been offset by one or is this brand new behaviour? Has it ever done it with the 3.40 firmware?

robxbl69, 2023-07-13:
This is new behaviour AFAIK, haven't seen this in 3.40 before

robxbl69, 2023-07-13:
This is weird as I just powered off and back on again and the display is okay now :-/ Maybe the issue was/is only there after I updated from 3.40 to the new version and didn't power cycle?

keirf, 2023-07-13:
Could be a reset / power on issue as you also saw garbage at least once?

keirf, 2023-07-13:
I was unable to reproduce corruption even if I repeatedly reset the Gotek but don't power cycle. This is as expected as FlashFloppy resets the display itself using a standard sequence. The off by one shift you saw is a settable configuration of the LCD controller, however FlashFloppy never configures it so something must have gone awry during reset for the configuration to get messed up.

closed 2023-07-14 by keirf

keirf, 2023-07-14:
I also wasn't able to repro erratic scrolling. If you do see that again, that could be interesting.

keirf, 2023-07-14:
If you do want to futz around with it some more, here's a link to the very latest build:
https://github.com/keirf/flashfloppy/actions/runs/5551443591

Actually you can always also find the latest build under the Actions tab and look for the latest CI run on branch `master`.