keirf/flashfloppy #810: GOTEK SFRKC30.AT2 with AT32F415KBU7-4MCU QFN32

BLUEGALAXY75, 2023-07-14:
Hello,

I just bought a GOTEK SFRKC30.AT2 (AT32F415KBU7-4MCU QFN32) with FLASHFLOPPY 3.39 flashed for my Amiga 1200. 
It boots up and USBs keys are fully detected. However, I did not succeed making Amiga read any of my adfs.
(Symptoms: The tracks counter increments rapidly up to 255.0 but the adf is not read)

I flashed The IO-test (3.39 and 4.7a versions) to check the IOs of the chipset and observed that the following pins are stuck reading as LOW, even if 5V is applied:
- LCD 1: Drive Select B
- LCD 2 (floppy-16): Motor On
- LCD 3 (floppy-18): Direction 
- LCD 7 (floppy-32): Side/Head Select
(numbers 1, 2, 3 and 7 never appear)
Any idea about this issue would be much appreciated.
Thanks by advance

keirf, 2023-07-14:
Drive Select B and Motor On are unconnected on AT2 and will always read as LOW.

Pins 18 and 32 should work. So, perhaps the Gotek is broken?

BLUEGALAXY75, 2023-07-15:
Thank you for your reply and for your complement of information regarding the "drive select" and "motor on" pins .
I'm afraid you are right: perhaps my Gotek is broken :-/

closed 2023-07-17 by keirf