keirf/flashfloppy #811: Windows won't recognize Gotek SFRKC30.AT3

bromine3, 2023-07-15:
I have read and re-read the instructions but I still cannot get Windows to recognize my Gotek. Any suggestions?

keirf, 2023-07-15:
Jumpers at S1 and JC should be sufficient. Along with a suitable image file on your USB drive (eg. a 1.44MB IMG file).

bromine3, 2023-07-15:
So I believe I have the jumpers in the correct place. I'm using the usb-a to usb-a method but still I cannot get the Artery software to see my device.

keirf, 2023-07-16:
I see. The problem is to get the Gotek recognised as a usb dfu device. At the very least you should here the device hotplug chime when inserted, Windows should look for a driver, device should appear in device manager under usb devices, etc. If that doesn't happen then you misjumpered or the Gotek is bad or quite possibly your cable is bad.

bromine3, 2023-07-16:
Ok, thank you for the insight. Are there any gotek models that you would recommend?

keirf, 2023-07-16:
I personally recommend the AT32F435 based models, which have PCB model number SFRKC30.AT4.35. See https://github.com/keirf/FlashFloppy/wiki/Gotek-Models

As a bonus these will generally be ready flashed with FlashFloppy, as there is no other "factory firmware" for this model.

However, this doesn't really have anything to do with your current problem. Your Gotek absolutely should identify as a programmable device to Windows.

closed 2023-07-17 by keirf