keirf/flashfloppy #812: Reading Disks off GOTEK with STM controller and 3 digit display

DjAlmadaPT, 2023-07-18:
Hi All,

I have a GOTEK with the STM controller model SFRC9220.
Almost 2 years ago, I have flashed the Flashfloppy bootloader and firmware (when no usb stick is in the drive it shows F-F).
I was able to put some floppies there and it worked, however, due to recently being a father I have abandoned it and wanted to pick it back up now.

I am trying to build an old MSDOS/Windows 9x system based on an ABIT PX5 Motherboard, Intel Pentium CPU and 128MB of SDRAM (I know it's overkill but that's what I had lying around) and the problem I am having is that, the disks start loading, there's activity with numbers flashing on the GOTEK screen, but a few seconds later, it just reverts to 000 which is the position I have the MSDOS 6.22 Disk 1 mapped to.

What have I done so far:
-Formatted the USB stick as FAT 32
-Upgraded both the bootloader and the firmware to the latest version 3.41
-Loaded the FF.CFG file and changed the following settings (interface=ibmpc and jc, host=unspecified and pc-dos, nav-mode= indexed)
-Renamed the MSDOS 6.22 disks 1, 2 and 3 to DSKA0000_msd622d1, DSKA0001_msd622d2, DSKA0002_msd622d3 to assign them to the positions 000, 001 and 002 on the drive

I am attaching a picture of my GOTEK board for reference, however, jumper wise I have S1 closed (as per indication on the wiki to be compatible with most IBM PCs and I also have a piezo speaker on JB.

Appreciate any help you might be able provide!

Thanks in advance!

[image]

DjAlmadaPT, 2023-07-18:
> It sounds like some progress. Try the simplest possible configuration:
> 
> 1. A FAT32-formatted stick with only _one_ file on it. A bootable disk image. It should be 720k or 1.44M in size, precisely, with extension .IMG
> 2. Jumpers at J1 and JC only
> 3. Reset any cached configuration by holding both Gotek buttons when no stick is inserted, until the display changes to "rst".

Thanks for coming back to me Keirf.
I am assuming when you meant J1 you mean S1? Correct?

keirf, 2023-07-18:
Yes, sorry: S1 and JC only.

DjAlmadaPT, 2023-07-18:
@keirf , done that and the result are the same. 

Got a different USB stick, formatted as FAT32.
Loaded the .IMG and FF.CFG where the only settings changed were interface=ibmpc, host=unspecified and nav-mode=indexed.
Same results, numbers started popping on the LCD, saw on the screen "Starting MS-DOS" and then back to 000, computer frozen.

Question now is and sorry if I didn't understand correctly, should I keep the FF.CFG inside along with the IMG or ONLY the IMG I want to load from?

I've also tested without the FF.CFG inside, just the IMG and it just hangs on BIOS post (image below)
[image]

keirf, 2023-07-18:
Just one IMG file on the USB stick. No FF.CFG. Reset flash configuration by holding both buttons on the Gotek while no stick is inserted, until display changes from F-F to rSt. 

IMG file needs to be exactly 1.44MB since your drive A is configured as 1.44M in BIOS. Gotek needs to be positioned after cable twist on your ribbon cable.

keirf, 2023-07-18:
Your best bet is to fit an OLED display. Fairly easy to do. Then you don't need indexed mode!

closed 2023-07-19 by keirf