keirf/flashfloppy #814: SP1200 save file corrupts when Sync is set to MIDI on the SP1200

rCket, 2023-07-27:
SP1200 save file will not load if Sync is set to MIDI when you perform a save. It gets stuck/frozen at Track 12.1.

Without Sync set to MIDI on the SP1200 all saving and loading appears to work fine.

Using SFRKC30.AT4.35
Only jumper set is S0

If you save on the SP1200 whilst having Sync set to MIDI:
When it saves it does a faster skip over tracks 12, 13.
When you try and boot with that image file it stops at track 12.1 and the display reads "NoDataTransfered".

If I boot from a working image file, then select the corrupted image file (created above), I can load the Sequence data from the corrupt image file but the Sounds data will not load. Not even individual sounds on pads.

Same results from firmwares 3.37, 3.38, 3.39, 3.40, 3.41, 4.7a.
Used a Sandisk Cruiser Fit 16GB.

FF.CFG
"
interface=shugart
display-type = oled-128x64-rotate
"

Without Sync set to MIDI, the drive saves and loads fine with all the firmwares I used above.

I also tested a Lotharek HxC Rev F emulator and that saved and loaded fine with Sync set to MIDI.

keirf, 2023-07-27:
Interesting that it's even with 435 chip and even on the 4.x branch. How odd. A few options you could mess with, one by one, in FF.CFG (revert each before trying the next):
```
index-suppression = no
write-drain = realtime
track-change = realtime
```

You can find the defaults for each of these, to revert back to, in `examples/FF.CFG`.

keirf, 2023-07-27:
Also try another USB drive if you can, though of course a SanDisk Cruzer Fit should be totally fine.

rCket, 2023-07-27:
Will try and find a different USB stick but I have a house full of Sandisk Cruzers, lol

keirf, 2023-07-27:
I'll ask if anyone else I know sees this issue on a SP1200. Short of getting a logic analyser on the floppy interface it may be hard to work out what's going on. A logfile firmware dump of a write attempt could be useful actually https://github.com/keirf/flashfloppy/wiki/Logging-To-USB-Drive

The USB drive is clutching at straws really. Since you see a difference in emulated behaviour (tracks are skipped quickly) it doesn't really sound like it might be a write-back-to-usb problem. It's just an easy thing to try if you do have a different stick.

rCket, 2023-07-27:
** FlashFloppy 4.7a
** Keir Fraser <keir.xen@gmail.com>
** github:keirf/flashfloppy

Build: Dec  4 2022 14:21:37
Board: Standard
Config: Flash Slot 5 (ver 2, size 73)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: OLED found at 0x3c
OLED: SSD1306
Display: LCD/OLED

> Device speed: Full
 VID : 0781
 PID : 5571
> Class connected: 08 (MSC)
 Manufacturer : SanDisk'
 Product : Cruzer Fit
 Serial Number : 03023511121021040434
Cache 718 items
0:D: 'PROJECTS'
Cache 718 items
1:D: '2023'
Cache 718 items
2:F: 'DUNK.hfe' 
Cache 718 items
Mode: Native
Current slot: 2/6
Name: 'DUNK' Type: hfe
Attr: 20 Clus: 0000c752 Size: 2540544

** FlashFloppy 4.7a
** Keir Fraser <keir.xen@gmail.com>
** github:keirf/flashfloppy

Build: Dec  4 2022 14:21:37
Board: Standard
Config: Flash Slot 5 (ver 2, size 73)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: OLED found at 0x3c
OLED: SSD1306
Display: LCD/OLED

> Device speed: Full
 VID : 0781
 PID : 5571
> Class connected: 08 (MSC)
 Manufacturer : SanDisk'
 Product : Cruzer Fit
 Serial Number : 03023511121021040434
Cache 718 items
0:D: 'PROJECTS'
Cache 718 items
1:D: '2023'
Cache 718 items
2:F: 'DUNK.hfe' 
Cache 718 items
Mode: Native
Current slot: 2/6
Name: 'DUNK' Type: hfe
Attr: 20 Clus: 0000c752 Size: 2540544
Fast Seek: 1 frags
Cache 718 items
Before: After: Current slot: 3/6
Name: 'DUNK_000' Type: hfe
Attr: 20 Clus: 0000d448 Size: 2540544
Fast Seek: 1 frags
Cache 718 items
Before: After: Current slot: 2/6
Name: 'DUNK' Type: hfe
Attr: 20 Clus: 0000c752 Size: 2540544
Fast Seek: 1 frags
Cache 718 items
Before: After: Current slot: 3/6
Name: 'DUNK_000' Type: hfe
Attr: 20 Clus: 0000d448 Size: 2540544
Fast Seek: 1 frags
Lengthy write. 512 bytes took 35547 us
Lengthy write. 512 bytes took 10404 us
Lengthy write. 512 bytes took 36025 us
Lengthy write. 2048 bytes took 20905 us
Lengthy write. 512 bytes took 33392 us
RDATA underrun! 0-84-200
RDATA underrun! 0-84-200
RDATA underrun! 0-84-200
RDATA underrun! 0-84-200
RDATA underrun! 0-84-200
RDATA underrun! 0-84-200
Lengthy write. 2048 bytes took 189268 us
Lengthy write. 512 bytes took 9136 us
Lengthy write. 512 bytes took 35945 us
Lengthy write. 2048 bytes took 19222 us
Lengthy write. 512 bytes took 29023 us
Lengthy write. 512 bytes took 18178 us
Lengthy write. 512 bytes took 35137 us
RDATA underrun! 0-1aa-200
RDATA underrun! 0-1aa-200
RDATA underrun! 0-1aa-200
RDATA underrun! 0-1aa-200
RDATA underrun! 0-1aa-200
RDATA underrun! 0-1aa-200
Lengthy write. 2048 bytes took 186419 us
Lengthy write. 512 bytes took 9433 us
Lengthy write. 512 bytes took 35615 us
Lengthy write. 2048 bytes took 16639 us
Lengthy write. 512 bytes took 33457 us
RDATA underrun! 0-1cd-200
RDATA underrun! 0-1cd-200
RDATA underrun! 0-1cd-200
RDATA underrun! 0-1cd-200
RDATA underrun! 0-1cd-200
Lengthy write. 2048 bytes took 186777 us
Cache 718 items
Before: After: Current slot: 2/6
Name: 'DUNK' Type: hfe
Attr: 20 Clus: 0000c752 Size: 2540544

rCket, 2023-07-27:
[FFLOG-DUNK_000-corruptedSave.TXT](https://github.com/keirf/flashfloppy/files/12184669/FFLOG-DUNK_000-corruptedSave.TXT)

The DUNK.hfe file is the good, uncorrupted file.
I load that and then turn on 'Sync MIDI' on the SP1200, then switch to the DUNK_000 file and save from the SP1200 to the DUNK_000.hfe image.
This is the log file for that.

rCket, 2023-07-27:
I'll upload a save-log of a save without turning Sync MIDI on.

rCket, 2023-07-27:
[FFLOG-DUNK_000-goodSave.TXT](https://github.com/keirf/flashfloppy/files/12184754/FFLOG-DUNK_000-goodSave.TXT)

The DUNK.hfe file is the good, uncorrupted file.
I load that and then switch to the DUNK_000 file and save from the SP1200 to the DUNK_000.hfe image.
This is the log file for that.

This saves and loads fine.

keirf, 2023-07-27:
Can you grab logs with v3.41?

keirf, 2023-07-27:
Also please can you upload DUNK_000.hfe

keirf, 2023-07-27:
And finally, any reason you can think of that Sync to MIDI would affect save operations? It should be a totally unrelated setting right?

rCket, 2023-07-27:
Will do logs with 3.41 in a bit.
and upload DUNK_000.hfe, the corrupted version.

I have no idea why Sync to MIDI does this, yet.

keirf, 2023-07-27:
Upload a good HFE and a corrupt HFE, actually. That would be good to compare

rCket, 2023-07-27:
[FF-SP1200-SyncToMIDI-issue814.zip](https://github.com/keirf/flashfloppy/files/12185466/FF-SP1200-SyncToMIDI-issue814.zip)

keirf, 2023-07-27:
Both HFE images look fine in HxC floppy viewer. Perhaps some tracks get missed entirely and contain old sector data?

One thing I see in the SyncToMIDI log file is:
```
Write 28672-31743 (3072)... 5473 us
Write 0-4095 (4096)... 7516 us
Write 4096-8191 (4096)... 6530 us
Write 8192-12287 (4096)... 7881 us
Write 12288-16383 (4096)... 6446 us
Write 16384-20479 (4096)... 39832 us
Write 20480-24575 (4096)... 249610 us
Write 24576-28671 (4096)... 407630 us
Write 28672-31743 (3072)... 230138 us
```

Those write times are monstrous. 400ms to write 4kB! So it is definitely worth trying a different sort of USB drive.

rCket, 2023-07-27:
Tried a couple of other USB sticks and it's the same story, stopping at 12.1.

rCket, 2023-07-27:
Might have to go back to the HxC Rev F until there's a FF fix. Problem is, I use MIDI sync all the time and it's so easy to forget to turn it off before saving.
Love FF and the OLED and rotary and config options.
I'll be watching this space.
Cheers m8.

rCket, 2023-08-12:
Sorted. I think. Need to test everything loaded correctly but i think it should have.

side-select-glitch-filter = 100

Made it load in 1 minute. Faster than the older Gotek.

Still doing more testing, but, it's a breakthrough.

closed 2024-01-11 by keirf

rCket, 2024-01-11:
Hi Keir.Did you fix this in an update?I still have the exact same problem.Sent from my iPhone

Closed #814 as completed.

keirf, 2024-01-11:
No this is bug scrubbing. Without further information, such as a logic analyser trace, there is no way this will get fixed.

keirf, 2024-01-11:
That said it is worth giving v3.42 a quick try as I have just tweaked the handling of write data a little bit. Not specifically for this issue though!

rCket, 2024-01-12:
I’ll check it out, thanks