keirf/flashfloppy #815: IBM 2.88MB image even cylinders give sector error.

Kimontui, 2023-08-02:
Tested with versions 3.30 and 3.41. If you use a standard RAW IMG/IMA 2.88MB disc image. All even cylinders give sector seek error. First and all odd cylinders are readable. But all even cylinders give me sector seek errors if wrinting files. So Flashfloppy does not generate good sector index. If you surface scan with Scandisc/Norton Disc Doctor no bad blocks appear. So reading works.

Kimontui, 2023-08-02:
It is not a speed issue because img.cfg rpm=150 bps=250 with a 1.44MB drive and 2.8MB image gives the same results. But should work.

keirf, 2023-08-03:
Both sides of cyls 0,2,4,6,...?

Kimontui, 2023-08-03:
Gotek LCD displays active cylinders. In any 36 track floppy. The display will never show 2468 but will show 013579. And DOS COPY command only copies odd cylinders.

keirf, 2023-08-03:
A 36 track floppy?

Kimontui, 2023-08-03:
Few corrections. Sorry. 36sector per track. And it shows 013568. It does not write. But reads good. Tried gap2=41 in img.cfg and it did not work either. I can hack it to use 18 sect track and oversize the cylinders to 160. And modify floppy param block. But thats not standard.

closed 2024-01-11 by keirf