keirf/flashfloppy #818: Roland RA-95 - dEr error

andr229k, 2023-08-24:
I found a Roland RA-95 in a garage-sale and thought that it could be fun to give a try with Flashfloppy. The original drive is a Chinon FZ-357 [Commonly used for Amiga device too] and was configured as:

[image]
 
I would therefore conclude the correct jumper setting as **S0**. Here there is also a jumper on **MM**, I guess that related to motor?

I first tried to see how the device normally behave, when you turn on the RA-95 you can see that the LED is constantly on and I inserted a random disk and it gives me the error “dEr”, which means “Disk is protected”, so I execute a format-process on RA-95 and then it runs normal.

[image]

**Environment:**
Roland RA-95 with USB Floppy Emulator installed and Flashfloppy 3.40. Jumper setting on S0 due to information above from original drive. Interface = Shugart

**Symptoms:**
RA-95 pushed “dEr” error on screen as USB-stick with empty disk generated with HxC Floppy Emulator is inserted. Try to run the format procedure, experience that it randomly stops the formatting-procedure and shows “dEr”.

**Diagnose:**
•	The LED on the Gotek was constantly on, which was normal behaviour from the normal floppy drive too. 
•	That I’m able to initiate the format procedure, means that I’m on the right path.

**Actions:**
•	I tried both Shugart, IBMPC and JC. I found Shugart best behaving.
•	Generated a log [Attached]from a formatting attempt, where it manged to get to track 10.
•	After a few attempts I was able to successfully format an empty disk [attached].

Any suggestions or tweaks?

[FFLOG.TXT](https://github.com/keirf/flashfloppy/files/12432349/FFLOG.TXT)

[Blank.HFE.txt](https://github.com/keirf/flashfloppy/files/12432352/Blank.HFE.txt)

[image]

keirf, 2023-08-24:
Probably best trying an IMG file from HxC. Normally Roland is just a standard 730k or 1M44 track layout.

andr229k, 2023-08-24:
> Probably best trying an IMG file from HxC. Normally Roland is just a standard 730k or 1M44 track layout.

RA-95 dont really get into the formating process with .IMG compared to .HFE format, which is weird. I have attached the logs, which it provided for 2 attempts.
[FFLOG 1.44 MB IMG 2nd try.TXT](https://github.com/keirf/flashfloppy/files/12432771/FFLOG.1.44.MB.IMG.2nd.try.TXT)
[FFLOG 1.44 MB IMG.TXT](https://github.com/keirf/flashfloppy/files/12432773/FFLOG.1.44.MB.IMG.TXT)

keirf, 2023-08-24:
Attach an HFE that you have tried to format on this device.

keirf, 2023-08-24:
Oh is that what you already attached in the OP?

andr229k, 2023-08-24:
> Oh is that what you already attached in the OP?

Yes, but you need to rename the extension to .HFE as Github dont allow .HFE as extention.

keirf, 2023-08-25:
The format is PC 720k. So just use a 720k IMG file and format that. Or here is the IMG version of the formatted HFE that you posted originally:
[a.img.zip](https://github.com/keirf/flashfloppy/files/12436790/a.img.zip)

andr229k, 2023-08-25:
Interface as "jppc-hdout" did the trick, thank you!

closed 2023-08-25 by andr229k