keirf/flashfloppy #827: Write errors with "Iskra Delta Partner" Z80 computer (Intel P8272 FDC)

jjakob, 2023-09-14:
Hi. I'm using FlashFloppy, the latest version (3.41) with this computer. It's a 1985 Z80-based computer running CP/M 3 (Plus) with a Intel P8272 FDC. Originally it came with a Panasonic JU-465 drive with the Shugart interface. This drive can do 2-side DD MFM with 80 cylinders, 18 sectors of 256b per track.

I have some images that I downloaded that are correct and I can read them from FlashFloppy seemingly without errors.
They aren't the full 80-cylinder size, because the OS apparently only uses 77 tracks. I initially created a 80-track blank image, but I saw on the display that the format utility only went up to track 77, then finished, so from then on I used a 77-track blank image. Some images others made are only 73 track but the OS seems to also read them fine.

When I try to format a blank image (created with dd) the format write step finishes but the verify step reports tons of errors. Using hexdump I checked the resulting image and almost every 2nd sector is unformatted (full of 0x00 instead of 0xe5 which is what a formatted blank sector should be). The sectors that are formatted correctly are all 0xe5.

My FF.CFG is `pin02 = nc` (probably doesn't matter, pins 2 and 4 are NC on the CPU board).
IMG.CFG:
```
[idp::737280]
cyls = 80
heads = 2
secs = 18
bps = 256

[idp::709632]
cyls = 77
heads = 2
secs = 18
bps = 256

[idp::672768]
cyls = 73
heads = 2
secs = 18
bps = 256
```

I tried adding `interleave = 2`, `id = 0` or `file-layout = sequential` in all possible combinations to these options but they all error while reading or don't read at all from CP/M even when using the premade images. They also error when formatting immediately.  With the above configuration the images read in CP/M (I haven't yet tested reading large files, but `dir b: [full]` doesn't complain).
The documentation for this computer (and even the format utility, which is custom - not the standard CP/M utility) all say that the interleave factor is 2, but I seem to have the best success with leaving it at 1, I can't explain why.

I also tried adding `index-suppression = yes` but that didn't make any difference.

The image, incorrectly formatted: [new_77.idp.img.gz](https://github.com/keirf/flashfloppy/files/12610338/new_77.idp.img.gz)

I made Sigrok captures at 24MHz: [sigrok_floppy_format_bad.tar.gz](https://github.com/keirf/flashfloppy/files/12610407/sigrok_floppy_format_bad.tar.gz)
Unfortunately Sigrok has a bug that makes it impossible to save a session recording, but I exported it into raw bitstream files, and tgz'd them together with the pvs configuration, which you can then import into Sigrok in the import menu.

Other users of this computer report success with HxC. I don't know if they can write or just read, I'll ask. The attached screenshots are of the HxC config that they use.

Link to images others made: https://files.oddbit-retro.org/pub/Partner/BinaryDumps/Floppy/
Schematic (FDC is on the 2nd to last page): [ParnerCPUBoard.pdf](https://files.oddbit-retro.org/pub/Partner/Docs/Schematics/ParnerCPUBoard.pdf)

[HxC_screenshot_1](https://github.com/keirf/flashfloppy/assets/5466039/b2b5b672-876e-4688-874d-4a3cc435a86d)
[HxC_screenshot_2](https://github.com/keirf/flashfloppy/assets/5466039/f9a56d27-e0cb-4c56-87d8-0517eb0d41e7)

keirf, 2023-09-14:
Yes it is weird that interleave 2 isn't working so well for you, considering that's the HxC setting too.

Perhaps try formatting a blank double density HFE image file, and see if (a) that works; (b) either way you can attach it here and we can see how the host system likes to format its tracks at the raw level.

There's a blank double density image for download here: https://github.com/keirf/flashfloppy-images/blob/master/Unformatted/HFE/double_density.hfe (note there is a "download raw" link on this page, which is what you should use).

keirf, 2023-09-14:
Another thing would be to use a pre-formatted image and see if writing to it is reliable, rather than relying on formatting, which is going to be marginal with 256-byte sectors and an AT32F415-based Gotek which has only 32kB RAM. You can check RAM size on display of your Gotek if it has OLED, else press right button with no stick inserted and you will get a report of CPU speed, RAM size, and firmware version.

jjakob, 2023-09-14:
I tried writing files to a preformatted image (the Early.img from the link in my post) but also got errors when writing with pip with verify. `pip b:=a:cpm3.sys[vr` reported an error. It wrote something (I think it first copies to a file named `CPM3.$$$`, then verifies it, if it's ok, renames it). That's why I tried formatting, to see if there's an issue with writing, it would be easier to debug with a fixed write pattern that format does.

keirf, 2023-09-14:
Ok that's interesting the 435 Gotek shouldn't really have a problem. Please try formatting a blank HFE image file as I previously suggested. Even if it fails, the final contents of that file ought to be enlightening.

jjakob, 2023-09-14:
Here's the HFE image, created with HxC with the same parameters as above, but with interleave 2. Freshly formatted with lots of errors.
[formatted_bad.idp.hfe.gz](https://github.com/keirf/flashfloppy/files/12612240/formatted_bad.idp.hfe.gz)

keirf, 2023-09-14:
You don't need a preformatted HFE image, in fact it probably confuses matters. You are better to use the totally blank image I linked to in an earlier message.

jjakob, 2023-09-14:
I'm sorry, I missed that message. Here's a format run with that.
[double_density_formatted_bad.hfe.gz](https://github.com/keirf/flashfloppy/files/12612440/double_density_formatted_bad.hfe.gz)

jjakob, 2023-09-15:
Debug log from the serial port with firmware from commit 78f93ea
[debug_log.txt.gz](https://github.com/keirf/flashfloppy/files/12622339/debug_log.txt.gz)
This is from a different format run than the hfe image above.

keirf, 2023-09-15:
The output to HFE looks corrupted, as though there is a communications problem between host and drive. Perhaps the WDATA line is problematic? I wasn't able to work out how to open a meaningful Sigrok session. I tried running pulseview in Linux and then import the files, but the UI is pretty bad and there were no helpful labels on the traces (assuming I even imported the raw data correctly!).

I loaded the HFE images into HxC software and used the visual track analyzer. I can there see that sections of the write data look very odd (eg. sector size 8kB, cylinder 127, etc etc).

jjakob, 2023-09-15:
@keirf To import it into Pulseview:
- click on the down arrow next to the "Open" button
- select "Restore session setup" and open the ".pvs" file
- open the same menu next to "Open", select "Import raw binary logic data" and select the .raw.pvr file

The signal integrity of WDATA looks good to me. There is something weird going on with the DISK0 (DS0) line though. It's pulsing all the time even when the computer is idle and there is no disk access taking place. Then when the disk is being accessed, it pulses before and after each write cycle multiple times. I'll see if that is coming out of the 8272 like that or if there is something wrong with the logic that decodes it and drives the DS lines.
I saw that corrupt data in HxC too but I don't know what to make of it. It looks like the sectors are all recorded but their contents are wrong, like as if the corrupt data is being written by the DMA into the 8272. This uses DMA to write data into the 8272, so maybe something is wrong there.

jjakob, 2023-09-15:
I tried importing it now and there's a question about the sample rate and number of channels when importing the raw data. Set that to 8 channels and 24000000Hz (24MHz). Also there are no labels even if you restore the session from .pvs unless you're connected to a LA. So to see the labels you can open the .pvs file in a text editor and see them there.
Maybe you can use other software to view the raw data file, I don't know.

jjakob, 2023-09-15:
The 8272 scans all 4 drives by polling all of them every 2ms, so that pulsing of the DS line is normal for tha 8272. So if the data inside the sectors is wrong but the written pulses are there, that could mean that there's nothing wrong between FlashFloppy and the 8272 and it's either the 8272 or something between the 8272 and the rest of the system malfunctioning. There could be anything, I'd need to probe around with the LA. I could test it with a real floppy (I don't have one, I need to get it).

jjakob, 2023-09-16:
I found the issue. I probed the WD1691 with the LA, specifically the WDIN, EARLY, LATE and WDO pins. I found that sometimes WDO had 2 pulses where there should be one MFM write pulse, and sometimes they were too long (400ns instead of 200ns). Looks like the 8272 doesn't wait even 40ns (the time resolution of my LA) after changing EARLY or LATE states to sending the WD pulse. This violates the WD1691 timing requirement Wel (EARLY/LATE to WDIN) = min. 100ns. So I tried disabling write precompensation by pulling TG43 low. Format now ran and verified successfully!
It seems like even though this means that sometimes two pulses or one longer pulse are written to the floppy, a real floppy doesn't care (or it does alter the magnetic properties on the disk but not severely) and the 8272 can read back those malformed floppies because it only samples the first pulse in a MFM period and ignores the 2nd pulse until enough time has passed. Maybe it would be an improvement if FlashFloppy also did the same (ignored extra pulses)? Or somehow recorded and reproduced them as-is?

keirf, 2023-09-16:
FlashFloppy only cares about falling edges on wdata. Can you tell how close together are the falling edges of the double pulses? Also any idea which is the real edge (pulses should generally be 4/6/8us apart).

jjakob, 2023-09-16:
Pulse too wide (400ns instead of 200ns):
[image]
Pulse too wide and two pulses:
[image]
Zoom into two pulses:
[image]

Whenever there are two pulses, those are a combination of early and late. When there is a long pulse, it's a early+normal or normal+late. So always on a 200us grid (it's what this precompensation time happens to be adjusted at, there is a trimpot on the board to set it)

keirf, 2023-09-16:
Okay so it really doesn't matter whether we merge the pulse with preceding or following flux, or even ignore it entirely. On WDATA only the falling edges matter. All timing is between falling edges. And here we have pulses at either +200ns or -200ns, which is +/-10% of the bitcell clock. It's not near enough to push a bitcell outside the FlashFloppy timing window (1us to 3us).

The FlashFloppy flux processing for WDATA is very dumb. It expects accurately clocked data from a host, perhaps with precomp applied (which is never enough to push bitcells outside the timing window). There is no handling of "runt fluxes" at all. They will be processed as a 1 bit. Hence the problem here. It is easy enough to do something better with these short pulses. Hang on and I will make a test firmware for you to try.

keirf, 2023-09-16:
Please can you download and test the firmware linked under Artifacts on this page: https://github.com/keirf/flashfloppy/actions/runs/6208558773

You will need to be logged into GitHub to see the Artifact download link.

When you download you will get a zip. Update to the firmware contained in the non-debug zip file within, using the normal firmware update process.

jjakob, 2023-09-16:
Yes, that fixed it! I removed the jumper I installed (to enable write precompensation) and formatted with verification with no problem.
I don't exactly understand what the changed code does, it ignores the 2nd pulse?

keirf, 2023-09-16:
The new code incorporates the short flux into the following flux. It does this by moving directly to the next flux sample (aka the next falling edge of wdata) without updating the variable `prev`.

I also reworked the code a little bit and promoted some variables to int. So the patch is obscured by that a little bit.

Thanks for testing. This will be in the next v3.x release. You are safe to use the test firmware meanwhile.

closed 2023-09-16 by jjakob