keirf/flashfloppy #829: GOTEK-drive with Flashfloppy v3.39 in an Amiga 500 Rev. 6A gives green screen.

Joentje1978, 2023-09-21:
Gotek-drive with Flashfloppy v3,39 firmware works great at an Amiga 500 Plus Rev. 8 but when I connect this drive to an Amiga 500 Rev. 6A the Amiga won't start-up and gives a green screen.

I've upgraded the Flashfloppy-firmware from version v3.39 to the latest stable version v3.41 but no difference.

I'm 100% sure it's a Flashfloppy firmware issue.
GOTEK-drive has a Artery AT32F415KBU7-4_128K microprocessor.

keirf, 2023-09-21:
Why are you 100% certain that it is a firmware issue?

Joentje1978, 2023-09-22:
I think it's a GOTEK hardware issue in combination with Amiga revision Rev.6A OR it's a firmware issue. I think this problem can be solved by firmware. The Amiga 500 Rev. 6 works fine with a floppydrive and has no defects.

keirf, 2023-09-22:
Green screen immediately at power on is a RAM test failure. Not really possible for firmware to cause this. As you say, must be a hardware issue on the Amiga or the Gotek side.

closed 2023-09-22 by keirf