keirf/flashfloppy #832: flashfloppy df0 real floppy drive df1 not working

mabocrack, 2023-10-02:
FlashFloppy 3.41
SFRKC30.AT3

I have a gotek internal connected to floppy internal port of an amiga500 with kickstart 1.2.
When I connect an external real floppy drive to the floppy drive external floppy port it dosent work (it works with the original configuration with real floppy drive connected instead of the gotek).

Cheers

keirf, 2023-10-02:
Rather likely to be a hardware problem I think. For example, perhaps your external port doesn't work at all.

With the Gotek as DF0 and real drive as DF1, run Amiga Test Kit. Run the floppy test and check what drive IDs are detected. Then perform read and write tests on DF0 and DF1 (you will need to insert an AmigaDOS disk in DF1)

mabocrack, 2023-10-02:
> Rather likely to be a hardware problem I think. For example, perhaps your external port doesn't work at all.
>
> With the Gotek as DF0 and real drive as DF1, run Amiga Test Kit. Run the floppy test and [...]

This is the first thing I looked for. All is correctly working. Gotek doesn't assign the ID to the external floppy drive (i think). From gotek AmigaKit adf the external floppy drive is not detected but makes all the DF1 tests reported as passed.  I have jumper on S0

keirf, 2023-10-02:
Lack of ID is why Workbench doesn't detect DF1. How are you connecting the real drive externally? Are you perhaps using an adapter intended only for use with a Gotek?

mabocrack, 2023-10-02:
The external drive is similar to this:

[https://upload.wikimedia.org/wikipedia/commons/thumb/2/29/Cumana_CAX_354_Disk_Drive_for_Amiga_%28front_and_top%29.jpg/800px-Cumana_CAX_354_Disk_Drive_for_Amiga_%28front_and_top%29.jpg](url)

and inside there is a citizen osdc-29c

[https://torlus.com/floppy/forum/download/file.php?id=450&sid=601787cb4e97b3a9e524490ab01a8b8d](url)

closed 2023-10-05 by keirf