keirf/flashfloppy #833: Korg I30 issue

andr229k, 2023-10-04:
**Environment:**
Korg I30 with Gotek [SFRC992D/ARM STM32F105] and Flashfloppy version 3.41.
Jumper on JC and S1
interface = ibmpc-hdout
Host = akai < [due to information on https://github.com/keirf/flashfloppy/wiki/Host-Platforms#korg about .img]

**Symptoms:**
The LED on the Emulator is not giving any handshake at loading/saving nor formatting. [Which is a bit frustrating] Like there is no connection between Floppy Emulator and device.

**Diagnose:**
-	Korg I30 does not use the default 34-pin [see pictures of cables and schematics].
-	Floppy cable is okay, at it seem normal functional on the original drive.
-	Someone in the FlashFloppy group “halfy-confirmed” above as working.
-	Korg I30 can use both 2DD and HD disk [Seen in manual]

**Actions:**
-	Removing Jumper from JC [same issue]
-	Trying IBMPC as Interface [same issue]

Any ideas how to proceed?

[image]
[image]
[1.44 MB.IMG.txt](https://github.com/keirf/flashfloppy/files/12805283/1.44.MB.IMG.txt)

keirf, 2023-10-05:
Sounds about right to me. Pin 2 is not connected at the floppy drive, so `ibmpc` and `ibmpc-hdout` and even `jc` with JC jumpered should all be equivalent, giving the required DISK CHANGE on pin 34. And drive select is at pin 12, and that is indeed jumper position S1.

andr229k, 2023-10-05:
> Sounds about right to me. Pin 2 is not connected at the floppy drive, so `ibmpc` and `ibmpc-hdout` and even `jc` with JC jumpered should all be equivalent, giving the required DISK CHANGE on pin 34. And drive select is at pin 12, and that is indeed jumper position S1.

**below is an example where:**
Left side = What you see on disk-page, when original floppy-drive is inserted. [no disk in Floppy-drive but device is recogniced.]
Right side = What you see on disk-page, when Gotek or no drive is inserted. [ 'unformated' what means 'i dont find a device'!]

Are there any way to diagnostic, if something is not receied correctly on the pins?

[image]

keirf, 2023-10-05:
Are you testing the Gotek with no USB stick inserted, for now? This will be equivalent to a floppy drive with no disk inserted. It might make sense to get identical Gorg host behaviour between these two setups before proceeding.

There are five lines from the drive to the uPD72070 controller chip: INDEX, TRACK00, WRITE PROTECT, READ DATA, DISK CHANGE.
Assuming the Panasonic drive is a 1.44MB PC drive (which it appears to be) then when no disk inserted
* INDEX and READ DATA are masked (they should have no activity). This matches Gotek.
* TRACK00 should behave normally, as it will on Gotek.
* DISK CHANGE should be asserted (LOW), as it will be on Gotek when no disk image is mounted.
* WRITE PROTECT may or may not be asserted. Commonly it is (eg Atari ST uses this to detect disk changes). Gotek will assert this when no disk image is mounted.

So... there should be no reason to see a behavioural difference between an empty (no disk) floppy drive and an empty (no stick) Gotek.

keirf, 2023-10-05:
Is the Gotek known good?

keirf, 2023-10-06:
Stick an AT in there and see what happens. Preferably a 435.

andr229k, 2023-10-06:
Just a last check-up before switching to AT-chipset and dumping the ST into the bin. See close-up pictures below, are there any settings you would define as "odd" or root-cause leading to issue?

[image]

keirf, 2023-10-06:
Looks like bog standard PC configuration to me.

closed 2024-01-11 by keirf