keirf/flashfloppy #834: 64x128 OLED display

sirdavid2, 2023-10-12:
Would it be possible to add 64x128 OLED display with 90 degrees rotation?

keirf, 2024-01-11:
No plan for this I'm afraid. With the fonts used I think it would be quite ugly and not fit the display well.

closed 2024-01-11 by keirf

sirdavid2, 2024-01-11:
Sorry but I don't understand the fonts problem. 128x64 is supported (even with 180 degrees rotation), so 64x128 with 90 or 270 degrees rotation should look exactly the same with the same fonts.

keirf, 2024-01-11:
Oh I see. I thought you wanted an OLED viewable vertically rather than horizontally. For your usage just get a 128x64 OLED.

sirdavid2, 2024-01-11:
I already have two 64x128 OLEDS and I wanted to use them somehow. I thought that rotating by 90/270 degrees shouldn't be difficult to add since rotation by 180 already exists.

keirf, 2024-01-11:
It's much more difficult. 180 degree rotation is simply a command in the oled controller. I think 64x128 doesn't even use the same controller.