keirf/flashfloppy #836: Add config option to invert SD card detect signal

djtersteegc, 2023-10-16:
The common push-push Micro SD connector you see everywhere in China

e.g. https://jlcpcb.com/partdetail/gswitch-GT_TF003_H018502/C5155564

grounds the CD pin when the card is inserted, the opposite of what FlashFloppy wants.  Instead of using two more hardware components to invert it, I propose a config option in FF.CFG to invert the `sd_inserted(void)` check in firmware.

keirf, 2023-10-16:
On a new board the additional couple of components is not much of an imposition. Are you trying to retrofit an SD connector to a factory Gotek?

djtersteegc, 2023-10-16:
Custom board I'm designing to plug directly into a 50 pin header on an old Z-80 SBC reproduction I've made (Ferguson Big Board I).  Yes, I could fit the extra components, and this isn't a high volume solution where pennies matter, but still seemed like a software solution is usually preferred to a hardware solution in cases like this.

Is the concern tech debt and increased complexity going forward?

keirf, 2023-10-16:
Increased complexity, risk of regression, and yet another config option. For the latter it would be better if the polarity were at least encoded in pc15-12 pins somehow. But it's a tough sell compared with two cheap components already used elsewhere on the BoM.

djtersteegc, 2023-10-16:
Technically I'm using a 74LS07 instead of the 74HCT04/BSS138 combo, so one extra BOM item for the FET, but that's immaterial here.  Is this what you have in mind for the inverter?

[image]

keirf, 2023-10-17:
Yes that's right.

closed 2023-10-19 by djtersteegc