keirf/flashfloppy #843: A small issue when System Rebooting

iosoft, 2023-11-09:
I was testing  **Gotek+FlashFloppy** (720KB mode with 1GB Memory Card) with my old **Core2Duo** system.

**I noticed a small issue -**
When I fresh start the PC, no issue.... the system boots into A:
Now, if I reboot the system (either CTRL+ALT+DEL or press the RESTART switch) many times, it doesn't find the Floppy Drive nor able to boot into A:
"No Boot Device Found!"

Only solution is the shut-down the PC and star it again.
I am not sure, may be, the system restarts too fast before Gotek+FlashFloppy gets ready....

Any solution?

keirf, 2023-11-09:
Gotek (and floppy drive in general) does not see reset signal so FlashFloppy is ready throughout the reboot. So I'm not sure what's wrong. Is the drive type specified in BIOS or is it auto detected (somehow)?

iosoft, 2023-11-09:
> Gotek (and floppy drive in general) does not see reset signal so FlashFloppy is ready throughout the reboot. So I'm not sure what's wrong. Is the drive type specified in BIOS or is it auto detected (somehow)?

No not _Auto Mode_. Set as 720K Mode.
I will share some screenshots and a short video later.

keirf, 2023-11-09:
BIOS bug? I mean is it normal and tested to hit restart "many times"?

iosoft, 2023-11-09:
> BIOS bug? I mean is it normal and tested to hit restart "many times"?

I will test more and let you know with more details....

keirf, 2023-12-05:
Any further update before I scrub this bug?

iosoft, 2023-12-05:
> Any further update before I scrub this bug?

OOPS! I totally forgot.
Pls give me a day or two. I will post a video.