keirf/flashfloppy #844: Suggestion: Update Process

iosoft, 2023-11-09:
I have a **Gotek+FlashFloppy** (720KB mode with 1GB Memory Card) with default 8-Segment Display.

Pls do a small addition to your _Firmware Update process_, so that on successful firmware update, pls rename or delete the firmware file.
(only on successful firmware update)

This will help me to understand if the update process is successful or not, default 8-Segment Display is very limited.

Thanks for your awesome work.

keirf, 2023-11-09:
Press right button with no usb stick inserted. Information is displayed including firmware version.

iosoft, 2023-11-09:
> Press right button with no usb stick inserted. Information is displayed including firmware version.

COOL!

closed 2023-11-09 by iosoft