keirf/flashfloppy #848: Feature request: Custom sector numbers

hanz45ru, 2023-11-18:
There's a proposition to make custom sector numbering schemes, as it now possible with sector sizes. It will be useful in cases where  software protection based on very big sector number, and very useful for ISDOS bootable floppies for ZX Spectrum with BDI (it uses nearly usual 5x1024 track layout, but sector numbers are 1-2-3-4-9). And, of course, it can be used for interliving definition not only by factor.

keirf, 2023-12-12:
One place I really must support this however, is in the Greaseweazle tools. Which can be used for IMG<->HFE conversions with no need for Greaseweazle hardware.

hanz45ru, 2023-12-16:
> I'm almost inclined to force you to use HFE image files for these sorts of layouts.

I think this will work. Only one fact stops me - this will be periodical switching from HFE to TRD and back.
and, HFE images is bigger than TRD/IMG ones. Not so big problem, but still looks not good enough.

> A further problem would be custom numbering which is not in ascending numeric order in the IMG file.

And, in theory, custom numbers allows any order of numbers, is it?

hanz45ru, 2023-12-16:
> One place I really must support this however, is in the Greaseweazle tools. Which can be used for IMG<->HFE conversions with no need for Greaseweazle hardware.

Diggin' in my 5.25" floppies, I get something that definitely will not work after hfe -> img conversion. So, maybe, there is a good idea - switch to HFE, when it needed.

closed 2024-01-29 by hanz45ru