keirf/flashfloppy #851: 关于FF的写出到u盘的故障。

newease, 2023-11-24:
设备型号：SFRKC30.AT4.35
简要描述：写出数据到映像720k正常，1440k不正常。（映像文件来源Image Library）
故障现像：当主机向设备写入数据时。无法写入U盘，映像文件格式（RAW）。大小（1.44M）。oled屏有'w'写入标识闪烁 ，当前写入 磁道显示正常。如果用有指灯的USB读卡器  代替U盘。可以观察到读写灯有闪烁。但是 ，拿下盘后在电脑上查看数据完全没有改变。如果，映像文件是720K。就一切正常。包括格式化。也可以正常完成。
我试过好几个不用品牌的U盘包括读卡器都是一样 。
我试过3.xx 4.xx的固件都是样。也尝试过修改FF配置文件也无果。

newease, 2023-11-29:
我试着刷写了LOG的固件。我怀疑可能是主机的时钟不是很准确，或者写入的格式不一样。但读取一切正常。

 谢谢作者写了这么好用的固件。

#** FlashFloppy 69e2c54
** Keir Fraser <keir.xen@gmail.com>
** github:keirf/flashfloppy

Build: Nov 28 2023 11:55:53
Board: Standard
Config: Flash Slot 4 (ver 2, size 74)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: OLED found at 0x3c
OLED: SSD1306
Display: LCD/OLED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 090C
 PID : 1000
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : General
 Product : USB Flash Disk
 Serial Number : 04GQFHCHIAJ92MVA
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 720 items
0:D: 'OSP_U10'
Cache 719 items
1:F: 'U100BOOT408B.img' 
Cache 720 items
Mode: Native
Current slot: 1/2
Name: 'U100BOOT408B' Type: img
Attr: 20 Clus: 00004c98 Size: 1474560
Fast Seek: 1 frags
Cache 587 items
Cache 720 items
Before: After: Cache 719 items
Before: After: Cache 719 items
Before: After: Current slot: 21/22
Name: '1m44' Type: img
Attr: 20 Clus: 00006e5c Size: 1474560
Fast Seek: 1 frags
Cache 587 items
Cache 719 items
Before: After: Current slot: 21/22
Name: '1m44' Type: img
Attr: 20 Clus: 00006e5c Size: 1474560
Fast Seek: 1 frags
Cache 587 items
Cache 719 items
Before: After: Current slot: 21/22
Name: '1m44' Type: img
Attr: 20 Clus: 00006e5c Size: 1474560
Fast Seek: 1 frags
Cache 587 items
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: IDAM Bad CRC: 1b64, 226
IMG: IDAM Bad CRC: 1021, 14
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: IDAM Bad CRC: 353d, 8
IMG: IDAM Bad CRC: cfa2, 234
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: IDAM Bad CRC: 526b, 240
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
IMG: Bad Wr.Off: -12034
IMG: DAM Unknown
Cache 719 items
Before: After: Current slot: 20/22
Name: '0020' Type: img
Attr: 20 Clus: 00006049 Size: 1474560

keirf, 2023-12-05:
Not sure on this one I'm afraid. Can this host write to a real 1.44M floppy disk okay? Anyway, it looks most likely a cabling or host setup issue.

One thing you could try is a high density HFE image file eg. https://github.com/keirf/flashfloppy-images/blob/master/Unformatted/HFE/high_density.hfe

newease, 2023-12-05:
谢谢你的回复！我试一试这个IMAGE。
再次感谢！我是一个设备维修人员。这个固件帮我了的大忙。我再也不用带上成堆的设备。

keirf, 2023-12-05:
A further possibility: Use a cheap logic analyser (Saleae type) to take a trace of a disk write attempt. You could take a trace of: INDEX (8), RDATA (30), WDATA (22), WGATE (24), STEP (20), SIDE (32), and DRIVE_SELECT (10 or 12).

The HFE image file may yield interesting information first, however. You will need to format the blank HFE image on the host. It will quite possibly fail, but the result could be interesting.

Oh! One question: Do you know for certain that the host uses PC 1.44M formatting at high density? Some systems may support PC 720k for compatibility, but have a custom high-density format. In this case the HFE image may be especially interesting, as it will handle a custom high-density layout with no problem.

newease, 2023-12-05:
我只抓取了INDEX,WDATA,WGATE , RDATA。文件有点大压缩后有16m.需要用DSVIEW来打开。需要上传么。

keirf, 2023-12-05:
That is quite large. You would need to upload to a cloud drive and send a link. I don't know if that is possible from your location!

The original drive supports 300rpm and 360rpm rotation at high density. Are you sure the drive is strapped for 300rpm operation? In some configurations pin 2 is an input to the drive which selects between 300rpm and 360rpm.

Also, although the data sheet explicitly mentions 18 sectors and 1.44M capacity, this is not actually enforced. At 300rpm high density the original drive would accept other track formats with similar MFM bit timings. For example, 5*1kB sectors = 1600kB total capacity.

Can you look at your WDATA trace and measure the pulse widths (falling edge to falling edge). They may be multiple of 2 microseconds. Or if 360rpm operation is expected it could be multiples of 1.67 microseconds.

keirf, 2024-01-09:
Please see #861 which could be the same issue and which is working towards a fix. Are you interested still in this issue, and testing a fix?

closed 2024-01-11 by keirf

newease, 2024-01-25:
感谢！完美解决！！！