keirf/flashfloppy #852: Interface issues with DEC Pro 350

snhirsch, 2023-11-27:
I'm having good luck with emulation of an RX50 using two Gotek/FFs.  That's the good news.  The bad news is the bootstrap firmware on this machine fails the floppy subsystem test when disks are ejected.  With a real RX50 it's happy with empty drives (doors open or shut).  I can workaround by always having images loaded but that necessitates a bit of quick finger poking after power up.  Is there a setting I'm missing that might convince the system that, yes, drives are attached.  What is it looking for?

keirf, 2023-11-27:
Could be wanting ready on pin 34? Could try setting pin34=low in FF.CFG

snhirsch, 2023-11-28:
Hi, Keir.  After spending some time in the DEC technical documentation I have a theory as to what might be happening.  What happens with FF in the following situation:

1. No image inserted
2. Drive selected
3. Step outward by 80-tracks

A real RX50 steps outward to the stop and asserts track 0.  I have a feeling FF does not do so.  Can you clarify?  I believe the DEC self-test watches for track 0 to determine whether the drive is present.

keirf, 2023-11-28:
FF should do the same: step outward to track 0 then ignore further step requests and assert trk0 line.

keirf, 2023-11-28:
You should be able to confirm this with the logic analyser.

snhirsch, 2023-11-29:
Bear with me as I'm a bit puzzled.  How does the FF know how many track steps consitute return to 0?  If no diskette is inserted there is no geometry setting.   There are so many short glitches on the floppy interface signals that a logic analyzer trace fills quickly and shows nothing useful.

keirf, 2023-11-29:
FF always simulates a drive with 256 cylinders. It powers on at cylinder 0. If the host drives to cylinder >80 then 80 steps wouldn't get you back to 0. But unclear why this would differ when image inserted vs not.

keirf, 2023-11-29:
Probably not. Best to add that with a small external circuit based on dskchg and select and motor, wire-ored onto the index line? Especially as most Goteks ignore motor anyway.

dl6lr, 2023-11-29:
Am 29.11.2023 um 20:54 schrieb Steven Hirsch:
>
> I figured out what the DEC RX50 is doing that's different: With no diskette 
> inserted (door open or closed) it holds index true (low) when select and 
> motor-on are asserted. I'm thinking that's what the firmware is looking for. 
> Is there a possibility of adding this option to FF?
>
INDEX can be gated through the ready condition of the drive, but a lot of 
drives can be jumpered to output the index signal disregarding the drives 
state (i.e. TEAC FD-55FV via E0 jumper, this is the default) . This seems to 
be expected by the DEC host. When no disk is loaded, a 5.25" drive will output 
INDEX low as the light is passed with no obstruction.Message ID: 
***@***.***>

keirf, 2023-11-29:
Perhaps in that case it doesn't matter whether motor is asserted or not? When no disk is inserted, in this mode, index would be asserted, regardless of motor?

I believe I could implement that.

snhirsch, 2023-11-29:
The method @dl6lr describes may work, but is more liberal than the RX50.  By observation, the RX50 qualifies index with select and motor-on.

snhirsch, 2023-11-29:
> Probably not. Best to add that with a small external circuit based on dskchg and select and motor, wire-ored onto the index line? Especially as most Goteks ignore motor anyway.

I see what you are suggesting.  But unless I'm misunderstanding something the first step pulse clears dskchg, correct?  I'm almost positive the check for index occurs after the drive is homed.

snhirsch, 2023-11-30:
Ok, I breadboarded a circuit to do what you suggest.  Set chgrst = delay-1 on the Gotek.  Running on a floppy drive exerciser I can see it's doing the correct thing: with motor-on and select asserted and no diskette the index line stays asserted.  When I insert the image, it shows the expected index pulse.  But, unfortunately, the machine continues to complain about the floppy subsystem.  Starting to run out of ideas.  I love a mystery and will continue to chip at it.  Thanks for the help!

snhirsch, 2023-11-30:
Yes, the RX50 is an odd double-drive with a single motor and stepper, but two heads.  That "overlap" near the end is probably where SEL1 goes active.

snhirsch, 2023-11-30:
Definitely SEL1:
[image]

keirf, 2023-11-30:
Looks like rdata, same as index, isn't gated on ready. So you get noise from the read amplifier.

snhirsch, 2023-11-30:
That's what I'm assuming and by process of elimination the controller is probably expecting to see it.  Is there a convenient point in the Gotek to pick off something I can inject as noise on read line using select + chg to qualify?

closed 2023-12-01 by snhirsch