keirf/flashfloppy #855: Where is FF.cfg file?

alex3331, 2023-12-05:
Hi 
I need to rotate my OLED screen, so the advice is to modify FF.CFG file - but where is this file?

tenuke, 2023-12-05:
Make or copy such text-file into the root of the usb-stick.
Example ff.cfg is in the flashfloppy.zip

alex3331, 2023-12-05:
Tried both
display-type=oled-128x64-rotate
and
display-type=oled-128x64 -rotate
without any luck. Am I missing something?

keirf, 2023-12-06:
The former should work (no spaces on the line, between sub options). Perhaps filename is wrong or it's not a plain text file.

alex3331, 2023-12-06:
The file was in flashfloppy 3.41 distribution in examples folder. edited with Notepad

keirf, 2023-12-12:
This should certainly work: `display-type=oled-128x64-rotate` (no spaces).

If it didn't there is something wrong with the name, location, or filetype of your FF.CFG file.

keirf, 2023-12-12:
You may also try `display-type=oled-128x64` (in case your display is already rotated, and you need it unrotated).

closed 2024-01-10 by keirf