keirf/flashfloppy #858: MSX2 with Artery AT32F435 Gotek, does not read 360k images?

micv2, 2023-12-23:
Hello keifr, 

I converted my Philips VG-8235/22 MSX2 computer to 720k with Gotek Artery AT32F435. I have the latest firmware on it and set host=msx in the FF config. The 720k images finally work great. The 360k, on the other hand, doesn't. I don't know what to do now, do you have a tip for me? I also tried it with other Gotek Artery AT32F415, 360k doesn't work. Unfortunately I don't have a Gotek with an STM chipset to test. Maybe you have an idea, otherwise I'll leave it at 720K. 

Best regards and Merry Christmas 🎄

Paul

keirf, 2024-01-10:
Have you specified `host=msx` in your FF.CFG file?

keirf, 2024-01-10:
Attach your FF.CFG file so I can check it. With `host=msx` it ought to work, assuming your MSX expects 360k disks to be 80-track, single-sided.

micv2, 2024-01-10:
Ok 👍

[FF.zip](https://github.com/keirf/flashfloppy/files/13890311/FF.zip)

keirf, 2024-01-10:
The parameter needs to be lower case. `msx` not `MSX`. The firmware is quite fussy but also doesn't give syntax errors I'm afraid.

micv2, 2024-01-10:
Wow, it works! Thanks 👍

Hopefully other users don't make the same mistake as me 😉

keirf, 2024-01-10:
It happens. :)