keirf/flashfloppy #859: Trk XX: skip 21ms - why? 26pin sfrc2da

crazycusti, 2023-12-25:
Hello!

```
> USBH_USR_DeviceDisconnected
> USBH_USR_DeInit
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 3538
 PID : 0070
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : PQI
 Product : PQI USB Flash drive
 Serial Number : E1NQDYZP
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 104 items
0:F: 'Disk01.img' 
Cache 104 items
Mode: Native
Current slot: 0/0
Name: 'Disk01' Type: img
Attr: 20 Clus: 000020f4 Size: 1474560
Fast Seek: 20 frags
Cache 32 items
Trk 2: skip 21ms
Trk 4: skip 21ms
Trk 6: skip 21ms
Trk 8: skip 21ms
Trk 6: skip 21ms
Trk 4: skip 21ms
Trk 2: skip 21ms
Trk 0: skip 21ms
Trk 2: skip 21ms
Trk 4: skip 22ms
Trk 6: skip 21ms
Trk 8: skip 22ms
Trk 6: skip 21ms
Trk 4: skip 21ms
Trk 2: skip 21ms
Trk 0: skip 23ms
Trk 2: skip 20ms
Trk 4: skip 21ms
Trk 6: skip 21ms
Trk 8: skip 21ms
Trk 6: skip 21ms
Trk 4: skip 21ms
Trk 2: skip 21ms
Trk 0: skip 21ms
Trk 2: skip 22ms
Trk 4: skip 21ms
Trk 6: skip 21ms
Trk 8: skip 22ms
Trk 6: skip 21ms
Trk 4: skip 21ms
Trk 2: skip 20ms
Cache 104 items
Before: After: Current slot: 0/0
Name: 'Disk01' Type: img
Attr: 20 Clus: 000020f4 Size: 1474560``` 

i wanna install a sfrc2da-26pin gotek in my travelmate 4000m. previous floppy was a citizen w1d. after many hours to get it working, i give up here.. why all tracks will be skipped?

greets

keirf, 2023-12-25:
The tracks aren't being totally skipped. The message is to do with synchronising timing across tracks to make index pulses a regular distance apart. The skip message is informing when that is not possible, and how much time is skipped to the next index pulse.

You can only get that message when you have messed with FF.CFG values. Your best bet is to put everything back to defaults by removing FF.CFG and then resetting any cached values in the Gotek by holding both buttons for a few seconds while no USB drive is inserted (I would say until display changes, but you don't have a display!).

The only likely useful setting to explore is `interface=shugart` versus `interface=ibmpc`.

crazycusti, 2023-12-25:
hey

ive tried many hours with different configs or no config. with only interface=ibmpc or shugart the drive do nothing, no reaction on the gotek. 

only with pin2=dens, the floppy seek work, no error from bios and the trackcounter on the oled display "spin up and down". on all other optionflags in FF.cfg i had an dead connection between notebook and drive. in combination with pin2=dens (or low 0V) with other options, change nothing, all tracks become skipped. also the suggested delay-3 or 4 in FF.cfg wich i found on other issues here for the sfrc2d(a,b) change nothing :(

i have install a oled display for debugging and io/test.

tried flashfloppy from 3.29 to 3.41 and 4.7a. also updated the bootloaders.

the Travelmate 4000M is a very stressful notebook, rs232 is complex and not standard, ide is 504mb, bios is crappy, no wonder if the floppy controller is the problem, but w1d citizen is not an option for me.

crazycusti, 2023-12-25:
```
Build: Jul 14 2023 13:00:30
Board: Standard
Config: Flash Slot 2 (ver 2, size 74)
Interface: IBM PC (pin2=dens, pin34=rdy)
I2C: OLED found at 0x3c
OLED: SSD1306
Display: LCD/OLED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 3538
 PID : 0070
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : PQI
 Product : PQI USB Flash drive
 Serial Number : E1NQDYZP
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 104 items
0:F: 'Disk01.img' 
Cache 104 items
Mode: Native
Current slot: 0/0
Name: 'Disk01' Type: img
Attr: 20 Clus: 000020f4 Size: 1474560
Fast Seek: 20 frags
Cache 32 items
Cache 104 items
Before: After: Current slot: 0/0
Name: 'Disk01' Type: img
Attr: 20 Clus: 000020f4 Size: 1474560
``` 

/
```
interface = ibmpc
pin02 = dens
pin34 = rdy
```

keirf, 2024-01-10:
Unfortunately you're unlikely to get any more useful support from me. Debugging this sort of thing remotely is hard, it could be all sorts of things, even a bad cable.

crazycusti, 2024-01-10:
ive tried a another sony floppy with 26pin, which is in good condition but too big. it works but without a floppy disc its struggle exact the same on boot as the gotek with flashfloppy (fdc failure but seeks). with floppy disc it boot a suse linux 6.3 floppy in 8 of 10 tries.

i think the ready signal and timings? is a huge problem for the floppy controller on this machine, or the floppy controller is bad. so thanks for your help but i think the travelmate and gotek with flashfloppy is simply incompatible.

/closed

closed 2024-01-10 by crazycusti