keirf/flashfloppy #863: Encoder pin diagram misleading while using  "KY-40" encoder

Number1Tiim, 2024-01-12:
Under the /Hardware Mods "KC30 Rotary Encoder" section, a diagram is given:
  n/c     CLK     +5V
  SW      DT      GND
 (JK)    (J8)

I don't know if my KY-40 encoder is just different, but I did continuity checks and it appears to be the same as the Gotek factory encoder. (which you cannot purchase separately)

I had issues with not being able to use the push button, as well as having an awkward "skip one rotation" when I changed direction. (left would work, but when going right, I had to skip the first rotation click and then it would go right, and vice-versa)

I realized that GND and 5V were reversed, that solved the rotation issue. When using my meter on the push button, I got continuity on the top pin, labeled "n/c" and not the "SW" pin labeled so adjacent to (JK). Reversing these pins solved my issue of not being able to use the push button.

I guess that CLK and DT are reversed too, however these seem to work left/right just fine.

This might be my encoder? I bought it here:
https://www.amazon.com/HiLetgo-Degrees-Rotary-Encoder-Arduino/dp/B07WZF1TXX/ref=sr_1_4?crid=1KMFZ6OI4N5X3&keywords=arduino+encoder&qid=1705071238&sprefix=arduino+encoder%2Caps%2C75&sr=8-4

Just posting this to help others who may have bought a similar encoder.

Thanks!

keirf, 2024-01-12:
Are you saying that you think the wiki description of pins at the Gotek board are wrong? Or that you had confusion connecting them to your specific rotary encoder?

Number1Tiim, 2024-01-12:
Dear @keirf,

Now that I look at it again, your diagram is correct, of course. 

It was my INTERPETATION that got me all wrong!

I looked at the diagram and thought, "oh these [n/c, CLK, +5V] pins are the top pins" (of the 90-degree bend)... 

Never once thinking if I bend the 90 degree pins straight up in my mind... and not concentrating on what was "top vs bottom", I would have wired the thing correctly, the first time...

Boy is this embarrassing...

keirf, 2024-01-13:
Ah I see. Most boards shipped with no encoder also ship without those pins!

closed 2024-01-13 by keirf