keirf/flashfloppy #867: PS/2 can read 720k image but not 1.44Mb

DonLafferty, 2024-01-30:
I've got a PS/2 Model 30 286 that seems to be having trouble reading a 1.44 image. When I use an actual physical drive it seems to have a similar issue but I was hoping if I used the Gotek with FF that it would read either one.

keirf, 2024-03-19:
Not sure I'm afraid. I think the pin assignments on PS/2 floppy interface can be a bit different than normal. Maybe there's a density notification signal being missed?

DonLafferty, 2024-03-19:
There is a slight difference but it's mainly the 5V and 12V come in on the ribbon cable. As for sensing the 720 vs 1.44 I don't see any difference.

I was able to solve this by getting a PS/2 1.44 floppy to run configuration and then it reads either one okay.

closed 2024-03-19 by keirf