keirf/flashfloppy #868: AT435 GOTEK not auto changing Amiga ADF files in Workbench when selected

mds-au, 2024-02-01:
Hi - I have a new SFRKC30.AT4.35 Gotek drive that I want to use on my Amiga 500.
Amiga has a multi-rom boot device and I can select KS 1.3 or 3.1. or a diagrom.
Standard use is booting into WB3.2.1 from a GVP A500 HD+ with a BlueScsi V2 with a 3.1 Kickstart rom.
All works fine and I have no issues with another Gotek that I have that is an AT415 model.

The issue I have is with the AT435 device it won't "auto eject" and then mount a newly selected ADF file that I select via the rotary encoder (or when selecting with the buttons) it just remains on the initial ADF File that was mounted when Workbench was booted. 

When I use the AT415 device I don't experience this problem.

I have flashed the AT435 with the latest stable 3.42 with no success.
I have tried downgrading to older versions also with no success.

The AT415 worked fine with its original 3.28 firmware and also works fine with 3.42.

Any suggestions / advice appreciated.     Regards   Mark

keirf, 2024-02-01:
It means pin 2 isn't working properly. That's the disk change signal. Either a hardware problem or incorrect jumpers or configuration. Jumper at S0 only. No FF.CFG required. Reset cached config by holding both buttons for a few seconds with no usb drive inserted.

mds-au, 2024-02-01:
Hi Keir thanks so much for the prompt reply. I confirm that I just have jumper S0 only set and tried resetting the cache as suggested but the problem remains. I had FF.CFG for the oled display upside down fix and removed it once done. I also tried a different interface board and cable (as I am using the Gotek as external drive) and periodically use the DF0-DF1 switch to boot from the Gotek. Looking like it is a hardware issue with the Gotek :(.

closed 2024-03-19 by keirf