keirf/flashfloppy #871: Artery AT32F415 QFN32 pinout (32 pins) and rotary encoder, is it possible?

giobbino, 2024-02-16:
Hi, I've followed a tutorial about how to mount a rotary encoder, but when I've discovered it didn't work (worked just the rotary button, non the encoder) I started with the troubleshooting and discovered the video talked about 48 pins AT32F415 while mine has just 32 pins. So I searched the datasheet and I've found about the three versions (64 pins, 48 pins and 32 pins).

Now the question is: is there a way to use the rotary encoder with the 32 pins version, and - just in case - where should I connect the clock and data from the rotary encoder?

keirf, 2024-02-17:
The rotary encoder must be connected at the "KC30" header as described in the wiki here: https://github.com/keirf/flashfloppy/wiki/Hardware-Mods#kc30-rotary-header

giobbino, 2024-02-17:
Thank you so much! For some reason I totally missed that info   _(like my wife always says "you can't even find water in the ocean... tipical male" LOL)_

It works fine now!

closed 2024-02-17 by keirf