keirf/flashfloppy #874: Any interest in a at32F435 , OSHW design ?

fenugrec, 2024-02-25:
I'm aware of the OpenFlops project (stm32f105-based), but wondering if there would be any interest in a design using the more capable F435 part.
I'm thinking of something that could be shared on e.g. OSHPark and jlcpcb/pcbway for easy ordering (i.e. easier than playing the aliX/epay roulette and ending up with the wrong version...)

It's a fairly simple board, but is there already a reasonably-complete schematic somewhere ? If not, I could always sift through the code to find the pin assignments.

keirf, 2024-02-26:
The OpenFlops design already supports AT32F435 as a drop-in replacement for the STM32F105. However this isn't mentioned anywhere in the docs. 

Possibly I should mention it under "Gotek Models" and/or "Enhanced Gotek" wiki pages?

Making a schematic of the new factory models could be useful anyway. That depends how fun you find it to reverse engineer schematics. :) The pin-out variations are largely covered in the file header here: https://github.com/keirf/flashfloppy/blob/master/src/gotek/board.c

fenugrec, 2024-02-26:
Thanks for the reply.

> already supports AT32F435 as a drop-in replacement for the STM32F105

Ah good to know, I hadn't realized they were pin-compatible.

> Making a schematic of the new factory models could be useful anyway. 

Do you think they changed it in any significant way, if the new mcu is pin-compatible ?

I wonder why they went with discrete mosfets for certain signals instead of the obvious 74LS07 (hex buffer, open-drain outs)...

keirf, 2024-02-26:
The schematic didn't change very significantly. The buttons and rotary encoder move around a bit, is all.

Possibly the MOSFETs are cheaper and hardier? Board space isn't an issue.