keirf/flashfloppy #875: Disk change issues in DOS 6.22 with 3.42 on 3 devices

boohyaka, 2024-02-26:
Hi keirf,

I'm scratching my head. I have three different Goteks (1 SFRKC30AT3 and 2 STM32F105) all flashed to 3.42 in three different DOS 6.22 IBM PCs, that I have modded with rotary+oled+speaker, I have S1 and JC jumped on every one of them, and all 3 do not properly detect disk change. Am I missing something stupid?

I tried to add a ff.cfg file with:

interface=ibmpc
chgrst=delay-3

but it doesn't change anything. I've resetted default config on boot, even reflashed bootloader+fw 3.42 on one just to test it, to no avail.

If I select a first image then dir, it's fine, then if I change the image, wait a bit then dir again, I will keep getting the first image content off cache. Pressing CTRL-C then dir works, I get the new content immediately, but this is not a very good solution as I can't do that when installing multi-floppy games or apps.

Any idea or things to try? Thanks a ton in advance.

boohyaka, 2024-02-26:
[image]

pic if that helps.

keirf, 2024-02-26:
Remove FF.CFG and reset cached config. You shouldn't need an FF.CFG to get this working, and meanwhile it can only confuse things.

It would be interesting to see if you get changed behaviour if you jumper S1 only (remove JC): Then I would expect a "Not ready reading drive A" error.

Do you have a real drive and disks you can test on one of these machines with the same ribbon cable?

On the STM32 based Goteks you could try a much older firmware version and see if that helps. For example v3.20.

boohyaka, 2024-02-26:
Thanks for the answer.

- Removed FF.CFG and reset cached config: same behaviour
- Removed the JC jumper and left S1: same behaviour. Boots fine, no error at all. It's like JC does nothing.
- Normal floppy drive: works fine
- v3.20 on old devices: no difference

But in the meantime I found a working and viable workaround.

http://www.manmrk.net/tutorials/DOS/help/drivparm.htm

If /C is omitted, the drive is configured as not supporting disk-change signalling. So I added:

DRIVPARM=/d:0 /f:7

to my config.sys. (d:0 = drive a:, f:7 = 1.44MB)

And...everything works correctly. Tried on all 3 machines, and it's fine on all three. So at least I have a software workaround.

Some more info if that's useful:
- All drives are set as A:, after the twist
- Comp1: 386 DX-40, ISA IO card
- Comp2: 486 DX2-66, VLB IO card
- Copm3: Pentium 133, onboard IDE

If you're interested in the issue I'll happily run any test you require if it can help!

Cheers

keirf, 2024-03-12:
PS in the firmware programming guide you can skip the "disable protection" steps. And at the step of ticking "Verify after download" you would also want to change the Erase option to "Mass erase".

boohyaka, 2024-03-12:
oh btw - jumper JC made absolutely no difference on the pin 34 behavior on the AT drive.

OK thanks for the suggestion, I will try to reprogram it fully as instructed

closed 2024-08-23 by keirf