keirf/flashfloppy #877: Was working now not - need some assistance

StargeezerLabs, 2024-03-04:
Apologies if this is not the proper place to request assistance.
I am running the latest Flash Floppy on a Gotek with Artery AT32F-415CBT17 on a TRS-80 Model III with 48k RAM.
The Gotek is connected with the proper cable type from the Gotek to the external floppy edge connector.
Drive 1's internal cable is disconnected so the Gotek can be drive 1.  My understanding is you can not run the Gotek as drive #2 on a Model III so as to make a 3 drive system (if there is a way, I'd love to know to try once this problem is resolved).  Jumpers S1 enabled and all other disabled.  I added jumpers for the OLED and encoder. I went back and double checked my pins and no shorts on jumper points.

That being said, things were working and now they are not.  FF.cfg is on the root of the USB.  Only mod to FF.cfg is interface=shugart.

Same USB when working. I am using hfe files (converted from dsk with HXC Floppy Emulator).  The file shows on the OLED but says 000/000 and when the machine accesses the Gotek it quickly moves between tracks and seems to get stuck on 19 & 20?
I have tried different files, tried converting the same DSK again, tried reformatting the USB, tried removing the OLED and encoder to return the Gotek to stock configuration.  I do not have a spare cable. I have not tried a different USB.

Any thoughts on what is up? Cable gone bad perhaps?  What does 000/000 indicate.

Thanks all:
Mark

keirf, 2024-03-04:
000/000 means you have selected the first image file (000) of a total of 1 (max image file is therefore 000). It is the idle display mode when not being actively accessed/stepped by the host.

Hopefully you changed something between it working and not, that you can drill down into. Otherwise that's gonna be fun to debug I'm afraid.

closed 2024-03-04 by StargeezerLabs

StargeezerLabs, 2024-03-04:
Thanks for the input. :-)