keirf/flashfloppy #879: PC-98 HDI format support (not HDM)

windship, 2024-03-11:
There's lots of HDI format of PC-98 HDD image because of Anex86 emulator. I think you can find them easily from internet. 
I couldn't find any converter or information about this. FlashFloppy can access HDM format but no way about HDI.

Can you add HDI format support for next update of FlashFloppy? 
You can find some technical information about HDI format here:
https://www.pc98.org/project/doc/hdi.html

keirf, 2024-03-19:
Example HDI images would be useful.

windship, 2024-03-20:
[YUNO.zip](https://github.com/keirf/flashfloppy/files/14661664/YUNO.zip)
[Ultima Underworld 2.zip](https://github.com/keirf/flashfloppy/files/14661705/Ultima.Underworld.2.zip)
Please check!

windship, 2024-04-08:
Hello, any progress?

levrin, 2024-05-02:
Last I checked, HDI is for hard disk images, not floppies.

keirf, 2024-05-02:
Yes that is true, and FDI images are already supported. FlashFloppy cannot support HDD images so HDI cannot be supported.

closed 2024-05-02 by keirf

windship, 2024-05-02:
Hmm but HDM is hard disk image format also and FF support it right?