keirf/flashfloppy #883: Flashing via USB-A to USB-C cable does not work

dstolpmann, 2024-03-25:
Hi,
I tried flashing an Artery based GoTek with a USB-A (male) to USB-C (male) cable as suggested in the [wiki](https://github.com/keirf/flashfloppy/wiki/Firmware-Programming). However, my system did not recognize the GoTek. I tried different cables on a USB-C port of my system and via a USB-A (male) to USB-C (female) adapter, but none of the combinations worked. I finally managed to flash the GoTek via a USB-A (male) to USB-A (male) cable, which worked on the first try.

The wiki has been updated to mention the use of a USB-A (male) to USB-C (male) cable in [this commit](https://github.com/keirf/flashfloppy/wiki/Firmware-Programming/_compare/095561236a1e200b94aeed759ee5ff948cd368a1) based on issue #655.

I am wondering if this maybe only works on some systems or with some cables. According to [this StackExchange answer](https://superuser.com/a/1779948), the USB-C port has a way to detect whether it should supply power to a connected cable/device. I also tried to measure continuity through a USB-A (male) to USB-C (male) cable plugged into the USB-A (male) to USB-C (female) adapter and could not measure continuity on the +5V line, but on the other lines. So it seems my adapter blocks the +5V line, depending on the cable/device connected.

Did anyone else have success with using a standard USB-A (male) to USB-C (male) cable?

keirf, 2024-03-25:
You know what, I don't think it's supposed to work. The A end should normally always be "host" and supply bus power. And so an A-C cable should not have the resistor(s) required to tell the C end to supply power. When it works, it's probably because cable or C receptacle is out of spec. I should update wiki to mention this may not work.

keirf, 2024-03-25:
Wiki updated. Thanks for the report!

closed 2024-03-25 by keirf